raspberrypi/documentation #3143: Readability / completeness items for the Python on Raspberry Pi page

gitbls, 2023-10-03:
A couple of suggestions for the Python on Raspberry Pi page
(https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/os/using-python.adoc)

Readability improvement: The first mention of Python Virtual Environments is the section header containing the text. I found myself looking back to see where I missed what it is. I'd suggest adding a third paragraph to the first section that communicates that upstream changes to the `pip` command may affect usage of this command on Raspberry Pi OS, and the solution to this is to use Python virtual environments.  This may also cause a readability rethink of the "Notice" in the 2nd section.

This page would also benefit greatly from a section that discusses how to use env's in, for instance, cron jobs, systemd services, etc., that run python-based scripts. I don't know the answer to this, and since it's not documented here, I'll have to go ask Dr. Websearch about it.

lurch, 2023-10-03:
> This page would also benefit greatly from a section that discusses how to use env's in, for instance, cron jobs, systemd services, etc., that run python-based scripts. I don't know the answer to this, and since it's not documented here, I'll have to go ask Dr. Websearch about it.

IIRC the easiest way to do that is something like:
```
source /path/to/your/venv/bin/activate && python /path/to/your/script.py
```

aallan, 2023-10-03:
> Readability improvement: The first mention of Python Virtual Environments is the section header containing the text. I found myself looking back to see where I missed what it is. I'd suggest adding [...]

Moved the `IMPORTANT:` block up to the first section in  cbb5996d2c80ecacdf709900c6ee055cf1cff9cf. That probably suffices. The preferred route would be to have users install Python packages via `apt` and bypass `pip` entirely.

> This page would also benefit greatly from a section that discusses how to use env's in, for instance, cron jobs, systemd services, etc., that run python-based scripts. I don't know the answer to this, and since it's not documented here, I'll have to go ask Dr. Websearch about it.

This is out of scope for our documentation. This section is there for people who are asking themselves "why doesn't pip work on my Raspberry Pi?" We're not going to extend the documentation beyond that. Anything more complicated is general Linux documentation, and there are plenty of other sources on the web for that.