raspberrypi/documentation #3145: Document kmsprint changes on Bookworm

aallan, 2023-10-05:
On Raspberry Pi [1-4] you can run `kmsprint` on Bookworm, and it'll list all the display devices (HDMI0 / HDMI1 / DSI) that you have attached. But on Raspberry Pi 5 `kmsprint` will only display the first "card" that it finds, and there are multiple separate "cards" in the Pi 5.

So `kmsprint --device=/dev/dri/card1` might display the attached DSI screen, and `kmsprint --device=/dev/dri/card2` might print the attached HDMI screens. Unfortunately the card numbers aren't "stable", i.e. devices might move around, and there are separate cards for HDMI, DSI, DPI and VEC.

aallan, 2023-10-05:
Possibly we just reduce the section down to bare bones and point end users to the author's Git repo.

lurch, 2023-10-05:
`kmsprint` appears in a few places in the docs. On the `develop` branch I get:
```console
$ git grep kmsprint
documentation/asciidoc/computers/configuration/display-rotation.adoc:Note that the `--output` entry specifies to which device the rotation applies. You can determine the device name by typing `kmsprint | grep Connector` on the command line which will display information for all attached devices.
documentation/asciidoc/computers/os/graphics-utilities.adoc:=== kmsprint
documentation/asciidoc/computers/os/graphics-utilities.adoc:The `kmsprint` tool can be used to list the display-modes supported by the monitors attached to the Raspberry Pi. Use `kmsprint` to see details of the monitors connected to the Raspberry Pi, and `kmsprint -m` to see a list of all the display-modes supported by each monitor. You can find source code for the `kmsprint` utility https://github.com/tomba/kmsxx[on Github].
documentation/asciidoc/computers/os/playing-audio-and-video.adoc:kmsprint | grep Connector
```

lurch, 2023-10-05:
ping @6by9 in case he wants to provide additional info.

aallan, 2023-10-05:
> kmsprint appears in a few places in the docs. On the develop branch I get¬

That's only 3 entries, the middle 2 entries are the the section, and the title of that section.

aallan, 2023-10-05:
The "playing audio and video" one talks about the headphone jack, which doesn't exist on Pi 5 anyway.

aallan, 2023-10-05:
The "display rotation" one seems to have been updated for Wayland already. Perhaps there's an alternative to kmsprint we can use there. Ah well.

aallan, 2023-10-05:
There appear to be no more references. This looks like a 15 minute fix to avoid implying it works on Pi 5.

aallan, 2023-10-05:
@nathan-contino Can you toss this in the queue fairly high up, and get it fixed up on the `pi5` branch (**not** on `develop`). Thanks.

aallan, 2023-10-10:
Not sure how #3136 got linked there? 🤷‍♂️

aallan, 2023-12-10:
Never got done. Needs doing.

nathan-contino, 2024-02-12:
Still needs doing.

stevedavies-rp, 2026-08-27:
Fixed in upcoming edit.

closed 2026-08-27 by stevedavies-rp