raspberrypi/documentation #3146: Cameras: Synchronous capture on HQ: Circuit is wrongly drawn and will give wrong voltage!

njhollinghurst, 2023-10-06:
https://www.raspberrypi.com/documentation/accessories/camera.html#synchronous-captures
https://www.raspberrypi.com/documentation/accessories/images/synchronous_camera_wiring.jpg

The text says the 1.65V pull up voltage can be made by dividing 3.3V exactly by half with suitable resistors. But the image shows the divider wrongly connected to 5V.  The image should be corrected to use a 3.3V pin.

Also I wonder if the recommended resistor value (two 1K in parallel) is a bit on the low side and should maybe be increased?

njhollinghurst, 2023-10-06:
BTW was originally reported on the [forum](https://forums.raspberrypi.com/viewtopic.php?p=2141709#p2141709)

aallan, 2023-10-06:
@nathan-contino Can you do this one as an interrupt task as soon as possible, thanks.

lurch, 2023-10-06:
While you're there, it _might_ also be worth ensuring that any `v` used to signify volts is always written with an uppercase `V` (e.g. `1.65V` instead of `1.65v`) and changing `3V3` to `3.3V` ?

njhollinghurst, 2023-10-06:
As for the resistor value: Saying "equally high impedance ( > 1kOhm)", as we do now, seems underspecified. I have searched the IMX477 datasheet in vain for XVS IO current sink capability.  Pending a definitive spec and/or thorough testing, my best hunch is that the picture should show 2.2K or 3.3K colours.

This is less important than the over-voltage.

aallan, 2023-10-06:
Fixed on `develop`. Will get pushed to production with the Bookworm public release.

closed 2023-10-06 by aallan