raspberrypi/documentation #3153: Are there Boot Folder Contents changes?

aallan, 2023-10-10:
With the changes coming with Bookworm, is the content of the [Boot Folder](https://www.raspberrypi.com/documentation/computers/configuration.html#the-boot-folder) still correct?

For instance, it mentions `wpa_supplicant`, with Network Manager running the end user shouldn't be touching this file at all. 

Also, how have the binary blobs changed for Pi 5?

lurch, 2023-10-10:
IIRC the EEPROM on the Pi 4 means that the Pi 4 doesn't need a `bootcode.bin`, and the EEPROM on the Pi 5 means that the Pi 5 doesn't need a `bootcode.bin` nor a `start.elf` file. I believe the Pi 5 will load `kernel_2712.img` if it exists, but fall back to `kernel8.img` if it doesn't.

Contents of the boot partition on a 64-bit Bookworm image:
```
bcm2710-rpi-2-b.dtb       bcm2711-rpi-cm4s.dtb  fixup.dat         start4cd.elf
bcm2710-rpi-3-b.dtb       bcm2712-rpi-5-b.dtb   fixup_db.dat      start4db.elf
bcm2710-rpi-3-b-plus.dtb  bootcode.bin          fixup_x.dat       start4.elf
bcm2710-rpi-cm3.dtb       cmdline.txt           initramfs_2712    start4x.elf
bcm2710-rpi-zero-2.dtb    config.txt            initramfs8        start_cd.elf
bcm2710-rpi-zero-2-w.dtb  fixup4cd.dat          issue.txt         start_db.elf
bcm2711-rpi-400.dtb       fixup4.dat            kernel_2712.img   start.elf
bcm2711-rpi-4-b.dtb       fixup4db.dat          kernel8.img       start_x.elf
bcm2711-rpi-cm4.dtb       fixup4x.dat           LICENCE.broadcom
bcm2711-rpi-cm4-io.dtb    fixup_cd.dat          overlays/
```

Oh yeah, one of the other changes is that Bookworm now uses an initramfs file by default, by usage of the [auto_initramfs](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/config_txt/boot.adoc#auto_initramfs) keyword in `config.txt`.

So (AFAIK) a Pi 4 Model B would use `start4.elf`, `fixup4.dat` `config.txt`, `bcm2711-rpi-4-b.dtb`, `cmdline.txt`, `initramfs8` & `kernel8.img` during bootup; whereas a Pi 5 would use `config.txt`, `bcm2712-rpi-5-b.dtb`, `cmdline.txt`, `initramfs_2712` & `kernel_2712.img` during bootup. And for both platforms, it might load some stuff from the `overlays/` folder, depending on what you've got in `config.txt`.

aallan, 2023-10-12:
> I might also note that the docs for headless installation say dropping wpa_supplicant.conf into the boot partition (/boot/firmware now it seems) will autoconfigure wifi on firstboot. This is not the case. The firstboot just ignores that file and wifi does 'not' get configured automagically on firstboot.

Yup. That got missed, see #3164.

aallan, 2023-10-23:
> IIRC the EEPROM on the Pi 4 means that the Pi 4 doesn't need a bootcode.bin, and the EEPROM on the Pi 5 means that the Pi 5 doesn't need a bootcode.bin nor a start.elf file. I believe the Pi 5 will load kernel_2712.img if it exists, but fall back to kernel8.img if it doesn't.

So, yes. There are boot folder changes.

aallan, 2023-12-31:
Too fast `github-actions` bot, too fast!

closed 2024-01-16 by aallan