raspberrypi/documentation #3155: CM4/PiZero/Pi5 camera connector at documentation/accessories/camera.html#schematics

kbingham, 2023-10-11:
The Camera Hardware schematics (https://www.raspberrypi.com/documentation/accessories/camera.html#schematics) shows the pinout of the 15 pin camera connector, but not the 22 pin version used.

Could https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/camera/camera_hardware.adoc be updated to also reference the 22 pin connector pin-out please? (While this pinout is easy to find online, I would rather reference Raspberry Pi documentation than an external vendor)

aallan, 2023-10-11:
Ping @davidplowman and @naushir on this one.

lurch, 2023-10-11:
Not _exactly_ what you're looking for, but there's also https://datasheets.raspberrypi.com/cmcda/cmcda-schematics.pdf and https://datasheets.raspberrypi.com/cmcda/RPi-CMCDA-1P1.zip

aallan, 2023-10-12:
So I'm going to get the design team to work this diagram (and the existing 15-pin diagram) over so they look the same, and just a bit nicer as part of general "improve the quality of the diagrams and photographs in the documentation" push. But for reference here's the 22-pin connector.

[image]

lurch, 2023-10-12:
I'm guessing that's one of the MIPI CSI/DSI transceivers on the Pi 5? Just for the sake of completeness, do we also want the pinout of the 22-pin camera connector on the Pi Zero / Zero W / Zero 2 W ?

kbingham, 2023-10-12:
https://datasheets.raspberrypi.com/cmcda/cmcda-schematics.pdf seems to show a different 22-pin pinout than the diagram above. Do we have different pinouts for connectors that are physically the same across the range of Pi products? (That surprises me)

I.e. - I expect to see IO1 (PowerEn) on pin17 on a 22 pin connector?

lurch, 2023-10-12:
It looks like the numbering-scheme in the CMCDA schematics is simply "reversed" when compared to the diagram above? E.g. the diagram above has 3.3V on pin 1 and GND on pin 22, whereas the CMCDA schematic has 3.3V on pin 22 and GND on pin 1. And when you take that into account, the pinouts **do** seem consistent?

kbingham, 2023-10-12:
Yes, indeed - it does look reversed. Does that mean that cables on Pi5 are in any way reversed? Having used cameras on the Pi5 already - I expect not - but maybe that's something to clarify in the documentation.

I can confirm however that adding a camera to a Pi5 with a reversed cable (Terrifyingly I have one, and hit this by accident) will cause the Pi5 to fail to boot up at all (not even to the bootloader) so it gets pretty obvious quite quickly if it's wrong.

lurch, 2023-10-12:
I think we'd only recommend using official Raspberry Pi cables, which obviously don't have this problem :wink:

kbingham, 2023-10-12:
> I think we'd only recommend using official Raspberry Pi cables, which obviously don't have this problem 😉

Tell that to the pile of mipi cables I have on my desk ;-) haha

lurch, 2023-10-13:
Also, see page 3 of https://datasheets.raspberrypi.com/cmio/cmio-schematics.pdf

philipaxer, 2025-05-14:
Hi All,
I am running into a documenation issue on the 22pin CSI2 header. The voltage level of the I2C is implicitly documented as there are pullups towards 3.3V. However there is no indication what voltage levels the other IOs are (except the DPHY lanes). The Orin NX developer kit also uses a 22 pin CSI connector, propably "compatible" and they specifically specify 1.8V on GPIOs and 3.3V on I2C. 

The cameras i could find only one pin as ENABLE and feed it into regulators where the ENABLE pin does not have JEDEC CMOS levels. But that doesnt help me much.

[Orin NX Carrier Board Spec](https://developer.download.nvidia.com/assets/embedded/secure/jetson/orin_nano/docs/Jetson-Orin-Nano-DevKit-Carrier-Board-Specification_SP-11324-001_v1.3.pdf?__token__=exp=1747228793~hmac=8cc297756257d0cdc09c7ec0efb7923a8c13b8825f976dba262b9d8d52a063d1&t=eyJscyI6ImdzZW8iLCJsc2QiOiJodHRwczovL3d3dy5nb29nbGUuY29tLyJ9
)

lurch, 2025-05-14:
Perhaps the [CM5](https://datasheets.raspberrypi.com/cm5/cm5-datasheet.pdf) and [CM5IO](https://datasheets.raspberrypi.com/cm5/cm5io-datasheet.pdf) datasheets might have the detailed info you're looking for?

philipaxer, 2025-05-14:
The CM5IO has two CSI2: one is pull-up to 3.3V, the the other are GPIOs towards the high speed connector. Not sure what level comes out of that. The picture is incosistent to me. 

CM5 says 
> These pins are actually GPIO pins on RP1, CAM_GPIO0 is GPIO34 and CAM_GPIO1 is
> GPIO35

lurch, 2025-05-14:
ping @dp111

dp111, 2025-05-14:
1.8v should for the GPIO pin as it is an enable for a regulator

philipaxer, 2025-05-14:
In this case things are inconsistent, right? Sometimes it is 1.8V when driven from GPIOs. Othertimes is 3.3V if driven via pull-up.

It would be good to have some guidance in the documentation what to expect. I am building something that connects to this interface which is not 3.3V tolerant.

dp111, 2025-05-14:
Okay if you are building something that connects to a Pi then it is 3.3v
signal. It is alway from a Pi a 3.3v signal. I don't think we have any
inconsistencies.

philipaxer, 2025-05-14:
Could you point me to some document which specs that for my internal documentation? Thanks

dp111, 2025-05-14:
Nothing specific , but look at the CM5IO board and you will see for the MIPI port that doesn't have control of the GPIO pin it it pulled up to 3.3v to enable the regulator.

philipaxer, 2025-05-14:
Would it be possible to add a hard statement to the pi docs that specifies this? Alternatively I can draft it based on your comment and create a pull request.

lurch, 2025-05-14:
@philipaxer Just in case you're not aware: @dp111 is the chap who designed the CM5 and CM5IO boards 🙂  https://www.raspberrypi.com/news/meet-the-engineer-dominic-plunkett-on-compute-module-5/

philipaxer, 2025-05-14:
Thanks for the hint. The reason I am so annoying here is that the eco system around this connector is growing and these two GPIOs are not well defined in terms of "easy
to approach documentation". I am not questioning that the Pi ecosystem fits together, but I see that others (NVIDIA for instance) also adopted this pinout. It seems they use it differently when it comes to the GPIOs.

We (Axonne Inc) are considering to use the Pi connector not only for Pi systems but also for other platforms that claim to be compatible. Thus, I want to make sure interoperability is by design and not by accident.

It would be great to have some ground truth, ideally in the Pi ecosystem. I hope that makes sense, and in am not coming across to pushy.