raspberrypi/documentation #3163: Add wifi to eth network bridge to documentation

BarnOwlDown, 2023-10-11:
Hi there!
I will be nice and appreciated to have an how to about a network bridge from wifi to Ethernet.
I searched online but I found numerous way to do it, and some of them don't work.

lurch, 2023-10-11:
You mean like this? https://www.raspberrypi.com/documentation/computers/configuration.html#use-your-raspberry-pi-as-a-network-bridge
:slightly_smiling_face:

BarnOwlDown, 2023-10-11:
No, sorry for this misunderstanding. I mean a reverse bridge. My Raspberry is connected to internet through wlan0 and I want to share internet connection to ethernet.

The documentation show how to share Ethernet connection to wireless, isn't it?

lurch, 2023-10-11:
ping @nathan-contino

aallan, 2023-10-11:
What the OP is asking for is a mirrored but identical implementation to the example shown in the documentation. In this case you use wlan0 as the parent network in the bridge, and eth0 as the other network.

closed 2023-10-11 by aallan