raspberrypi/documentation #3168: Substitute mentions of raspi-gpio with pinctrl

aallan, 2023-10-12:
`raspi-gpio` appears in in documentation/asciidoc/computers/compute-module/cm-peri-sw-guide.adoc and documentation/asciidoc/computers/config_txt/gpio.adoc. It is deprecated, and we should now be using `pinctrl`.

See raspberrypi/bookworm-feedback#51 for details, currently waiting for the `pinctrl` package to be updated in the repo by @pelwell.

aallan, 2023-10-12:
Looking at where`raspi-gpio` gets [mentioned](https://www.raspberrypi.com/documentation/computers/compute-module.html#bcm283x-gpios) it's not exactly good documentation around using it in any case. We can do better.

aallan, 2023-10-12:
Should add a mention of `pinctrl` in the [GPIO section](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#gpio-pinout) of the documentation. Possibly in preference to the existing content?

lurch, 2023-10-12:
> We can do better.

Looks like big chunks of that page only mention CM1 and CM3, and omit CM4. (all the mentions of BCM283x are a bit of a giveaway - BCM2835 on CM1, BCM2837 on CM3, but BCM2711 on CM4)

aallan, 2023-10-13:
@XECDesign says, "_Looks like the pi5 version of the package got lost somewhere while the repos were being shuffled. The version in apt should be okay now_," so we should be good to go at this point.

aallan, 2023-10-13:
Fixed in 34cc579d2b818bb8cbe9e48abf609f0dc87190dd on the `pi5` branch (accidentally, and I can't be bothered to cherry pick onto `develop`).

closed 2023-10-13 by aallan