raspberrypi/documentation #3171: Update 7-inch Touch Display with information on wvkbd

aallan, 2023-10-12:
Turn the `NOTE` block in the [Touch Display documentation](https://www.raspberrypi.com/documentation/accessories/display.html) into an actual section talking about how to use an on-screen keyboard now we know that `wvkbd` [works under Wayland](https://github.com/raspberrypi/bookworm-feedback/issues/14#issuecomment-1760344457).

mukral, 2023-10-17:
wvkbd is not listed in the menue. Since I've working with touchscreen I cannot call it without a keyboard via execute. Sorry, I don't know how to inetgrate it in the menue. Can someone please post a way?
Thank you
Markus

aallan, 2023-12-17:
Still needs doing.

aallan, 2024-02-18:
Not so fast github-bot!

closed 2024-02-26 by aallan

nguyenlamlll, 2024-04-12:
Hi @aallan, I see that you merged #3540. But the `wvkbd` instruction does not seem working on my side. After I run the isntall command, um... what next? It does not work out of the box. No keyboard shows up when I touch the text editor. I'm using Rasp Pi 4 and the official 7" display. So, maybe something is missing?
Thanks.

nathan-contino, 2024-04-15:
@nguyenlamlll : great question! Unfortunately it looks like `wvkbd` doesn't pop up automatically when you click a text entry field. Instead, you need to manually invoke it by running the `wvkbd` command in your terminal... which is obviously a little hard to do when you don't have a keyboard attached!

There's a forum post with some workarounds (https://forums.raspberrypi.com/viewtopic.php?t=358147), including a script pinned to the desktop and a macro connected to the middle mouse click. For now, you might be able to use one of those approaches. I'll investigate internally and with the `wvkbd` folks to see if there's any way to get the automagic bits working in the software side of things.

nguyenlamlll, 2024-04-15:
@nathan-contino: Thank you very much for your quick update! I will head over there and have a look.

lurch, 2024-05-22:
@nathan-contino More info in https://github.com/raspberrypi/bookworm-feedback/issues/14 too.

MateuszKusmierz, 2024-10-14:
Hi. I'm using Wayland display but found out that wvkbd is not working for me. It requires manually starting via terminal which is crazy... Is there any on screen keyboard that works automatically when clicking on text area?