raspberrypi/documentation #3176: Anchor links not working in Firefox?

aallan, 2023-10-15:
There are reports that the anchor links, e.g. https://rptl.io/venv, for the section titles are not working in Firefox (although they do work fine in Chrome).

lurch, 2023-10-16:
I've also seen a slight oddity in Chrome where if I **am** looking at an anchor link in the middle of a documentation-page in a tab, sometimes if I go back to the tab after a long time of not looking at it, the browser will "refresh" that page and in doing so it'll then auto-scroll back up to the top of the page rather than staying at the mid-page anchor I was previously looking at.
However I wasn't sure if that was due to a change-in-behaviour of Chrome, or somehow related to #3117 ?

nathan-contino, 2023-10-16:
I noticed this a while back. Interestingly, it seems that I start out on the correct position on the page, then my browser scrolls back up to the top after initial load. Not sure if that helps at all.

PR #3185 (2023-10-17): Wait for element to load before scrolling
This should fix #3176
closed 2023-10-17 by aallan