raspberrypi/documentation #3177: RPi4/BCM2711 DMA DREQ for SPI3?

abqjln, 2023-10-15:
Cannot find info about accessing DMA DREQ (RX/TX) for SPI3, although I do see a solved note solving SPI4 and SPI5 [https://github.com/raspberrypi/documentation/issues/1903](url). I have this working, as well as SPI0 and SPI6.

Can we add (is there?) a DREQ (RX/TX) for SPI3 on the peripheral map?

aallan, 2023-10-15:
Please see the [BCM2711 Peripherals](https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf) book. This is our documentation around the BCM2711 processor, and we've published all we can at this time.

closed 2023-10-15 by aallan

lurch, 2023-10-16:
AFAIK the table in the "4.2.1.3. Peripheral DREQ Signals" section of that PDF is complete. And there can't be any "extra" DREQs that aren't listed in that table, because the relevant register-field isn't big enough!
[image]

abqjln, 2023-10-16:
Thanks-you guys are really responsive. I reverse engineered something that works in my application (just posted on raspberry pi forum) based on the extending the previous mux bit answer for SPI4 and SPI5. And some rebooting and reflashing...

(working)
DREQ 18: SPI1 RX / SPI3 RX, alternate when set bit 26 of PACTL_CS.
(suspected but untested in my code since I don't use it)
DREQ 16: SPI1 TX/ SPI3 TX, when set bit 26 of PACTL_CS.

BUT... I don't know what else might be using bit 26... No issues for me so far but yes I "own" the hardware and know this is not necessarily portable or multi-anything.

New to this documentation process, not sure where to go with this. But the above works for me--all 5 DMA SPIs (0,3,4,5,6) running using DMA (am not using 1 and 2).

Cheers!