raspberrypi/documentation #3178: Hardware PWM

hexpwn, 2023-10-15:
I cannot find information on the Raspberry Pi manual concerning Hardware PWM.

As far as I have found, there's a single bullet point (https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#more) which mentions:

> Hardware PWM available on GPIO12, GPIO13, GPIO18, GPIO19

lurch, 2023-10-16:
See chapter 8 of https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf (or chapter 9 of https://datasheets.raspberrypi.com/bcm2835/bcm2835-peripherals.pdf for models of Raspberry Pi before the Pi 4).

closed 2023-10-16 by lurch

hexpwn, 2023-10-16:
Hi @lurch, 

I have read the BCM documentation (for the 2835/7 - RPi 3) which is sparse in information. For example it gives no explanation on how to use the PLL's and just says the PWM clocks can be "varied by the clock manager"; and also doesn't give the addresses of important registers, like the PWM control register (yes this is fixed by erratas in other documents - no linked by RPi in 

I had no idea that the BCM2711 documentation would apply to the RPi 3 since I saw no mention of that as a source material elsewhere.

Yes; it is possible to find all the necessary information in other sources and code-bases, but I thought this would be the place to suggest improvements on the RPi documentation specifically (and in this case - specifically the Hardware PWM documentation).

Cheers

aallan, 2023-10-16:
Unfortunately that's all the information around the BCM processors that we can release at this time.

lurch, 2023-10-16:
> I had no idea that the BCM2711 documentation would apply to the RPi 3

It doesn't - the BCM2711 documentation only applies to the Pi 4 / Pi 400 / CM4 / CM4S.