raspberrypi/documentation #3205: Bookworm netboot not compatible with the vers= and proto= options listed in official netboot guide

trejan, 2023-10-19:
This was tested on a Pi 5. I followed the [official guide for netboot](https://www.raspberrypi.com/documentation/computers/remote-access.html#set-up-nfs-root).

The new initramfs setup uses the userland [klibc nfsmount](https://git.kernel.org/pub/scm/libs/klibc/klibc.git) to mount the root FS. The issue is that klibc nfsmount handles options differently than the kernel. It doesn't like the "vers=" option in cmdline.txt as it is [hardcoded to only accept 2 or 3](https://git.kernel.org/pub/scm/libs/klibc/klibc.git/tree/usr/kinit/nfsmount/main.c#n135) and doesn't recognise the "proto=" option. Having either of those two in cmdline.txt will cause boot to fail with an error message from nfsmount being repeatedly displayed.

Removing the proto= option and setting vers=3 in cmdline.txt will work but this downgrades to NFSv3. It appears to default to tcp anyway.

aallan, 2023-10-19:
> This was tested on a Pi 5. I followed the [official guide for netboot](https://www.raspberrypi.com/documentation/computers/remote-access.html#set-up-nfs-root).

The documentation has not been updated for Raspberry Pi 5 yet as hardware has not started shipping to end users…

> The new initramfs setup uses the userland [klibc nfsmount](https://git.kernel.org/pub/scm/libs/klibc/klibc.git) to mount the root FS. The issue is that klibc nfsmount handles options differently [...]

…but it sounds more like a Bookworm issue rather than a Raspberry Pi 5 issue however! Do you have the same issue on Raspberry Pi  4 hardware?

trejan, 2023-10-19:
Yeah. This is a Bookworm issue and applies to all models not just Pi 5.

I originally opened it in bookworm-feedback as I wasn't sure if it was going to be a documentation fix or whether it'd need a workaround. klibc nfsmount not accepting all options is a [longstanding unresolved issue in upstream Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=409272) and there are a couple workarounds listed there. Both will create a non-standard config but using the real mount command will significantly increase the size of the initramfs.

closed 2023-10-25 by aallan