raspberrypi/documentation #3207: Issue with playing commercial DVDs

kip-the-elder, 2023-10-20:
Some commercial DVDs that play fine on exactly the same setup on bullseye, don't play properly on bookworm. Somemost work just fine.

When you open the DVDs in VLC, the main menu, where there is one, works fine. But... when I try to play the movie itself, it's pix-elated and jumpy almost like you might expect if someone had spread honey on the disc. :roll: :lol:

I don't think that this is a VLC problem, because, Xine and KODI do exactly the same thing. (I installed Xine, handbrake  and KODI to test it) Exactly the same. If I try to rip a disk on bookworm with handbrake, The output is unwatchable and the same as if watched in Xine/Kodi/VLC . If I rip it on the Bullseye Pi, it then plays without any problem on the bookworm Pi.

If I swap the bookworm SD card out and replace with the bulleseye; everything works as it should do.

Setup is as follows.

Raspberry pi 400 with official power supply.
OS installed on Samsung EVOplus 64GB.
Logitech Wireless mouse
DVD drive is : Bus 001 Device 006: ID 1c6b:a223 Philips & Lite-ON Digital Solutions Corporation DVD Writer Slimtype eUAU108
The DVD drive is connected via a "Sarent" 4 port powered hub.
OS kept up to date

libdvdcss is installed as per

sudo apt install libdvd-pkg
sudo dpkg-reconfigure libdvd-pkg

I also tried installing it from github in case the repo wasn't up to date.
I also tried downgrading to the same version that I'm using on the working bullseye setup. (1.4.2-1)

Neither had any effect.

Apart from the OS, both setups are identical. It's the same physical machine. I'm just swapping the SD cards.

Other things I've tried

Rebuilt the OS again from scratch. Just the basic desktop, then just VLC to start with. No difference
Tried switching from Wayland to X11. No difference.
Tried the USB "usb-storage.quirks" bodge in cmdline.txt in case it was an issue with the USB..as it does seem noticeably slower to initially access the drive; though I didn't see why it would be seeing as nothing about the hardware has  changed. It made no difference as I expected.
Tried both setups on another 400s. Same thing...Works on Bullseye, not on Bookworm.
Compared the config.txt and cmdline.txt files to see if there was anything obvious. Nope. nothing..
Swapped the VLC config files from the Bullseye machine to the bookworm machine. No difference.

Some of those that don't work
Oceans 8
Red Road
Unthinkable
Mission impossible Rogue Nation

These work fine
Master and Commander
Mad Max Thunderdome
The Matrix
Men in Black 3

aallan, 2023-10-20:
This is not a problem with the documentation. You might want to file this against the [bookworm-feedback](https://github.com/raspberrypi/bookworm-feedback/issues) repo.

closed 2023-10-20 by aallan

kip-the-elder, 2023-10-20:
Ok...sorry..wetwear issue.