raspberrypi/documentation #3208: Splitting up the Raspberry Pi Hardware category

aallan, 2023-10-20:
We really need to look into splitting up the information in the [Raspberry Pi Hardware](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html) category into slightly more granular top-level categories. The page currently spans several topics. 

GPIO and the 40-pin Header
Voltages
Outputs
Inputs
More
GPIO pinout
Permissions
GPIO in Python
Schematics and Mechanical Drawings
Raspberry Pi 4 Model B
Raspberry Pi 3 Model B+
Raspberry Pi 3 Model B
Raspberry Pi 2 Model B
Raspberry Pi 1 Model B+
Raspberry Pi 3 Model A+
Raspberry Pi 1 Model A+
Raspberry Pi Zero
Raspberry Pi Zero W
Raspberry Pi Zero 2 W
Product compliance and safety
Flammability Rating
The Raspberry Pi Compliance Support
Powered by Raspberry Pi
Approved Design Partners
Frequency Management and Thermal Control
Using DVFS
Measuring Temperatures
Adding Heatsinks
Raspberry Pi 4 Boot EEPROM
Boot Diagnostics
Updating the Bootloader
Updating the EEPROM Configuration
Automatic Updates
Bootloader Release Status
Boot Diagnostics on the Raspberry Pi 4
Raspberry Pi Boot Modes
Special bootcode.bin-only boot mode
bootcode.bin UART Enable
Boot sequence
Raspberry Pi 4 Boot Flow
First Stage Bootloader
Second Stage Bootloader
Bootloader Updates
Fail-safe OS updates (TRYBOOT)
TRYBOOT_A_B mode
Raspberry Pi 4 Bootloader Configuration
Editing the Configuration
Configuration Properties
Configuration Properties in config.txt
Secure Boot configuration properties in config.txt
Updating to the LATEST / STABLE bootloader
USB Boot Modes
USB Device Boot Mode
USB Host Boot Mode
USB Mass Storage Boot
Raspberry Pi 4B and Raspberry Pi 400
Compute Module 4
Raspberry Pi 3B+
Raspberry Pi 2B, 3A+, 3B, CM3, CM3+, Zero 2 W
Booting from USB Mass Storage
Known Issues
Special bootcode.bin-only Boot Mode
Hardware Compatibility
Multiple Bootable Drives
Network Booting
Debugging Network Boot Mode
Known Problems
GPIO Boot Mode
Pin Assignments
Boot Flow
NVMe SSD Boot
Required Hardware
Required Software
Troubleshooting
HTTP Boot
Requirements
Keys
Secure Boot
Parallel Display Interface
GPIO Pins
Disable Other GPIO Peripherals
Configure a Display
General Purpose I/O (GPIO)
GPIO Pads
Interrupts
Alternative Functions
Voltage Specifications
GPIO Pads Control
Peripheral Addresses
Industrial use of the Raspberry Pi
One-Time Programmable Settings
Write and Read Customer OTP Values
Locking the OTP Changes
Making Customer OTP bits Unreadable
Device specific private key
OTP Register and Bit Definitions
OTP Registers
Power Supply
Typical Power Requirements
Power Supply Warnings
Back-powering
Serial Peripheral Interface (SPI)
SPI Hardware
SPI Software
Troubleshooting
Universal Serial Bus (USB)
Maximum Power Output
Raspberry Pi 4
Raspberry Pi Zero, 1, 2 and 3
Known USB Issues
Raspberry Pi revision codes
Old-style revision codes
New-style revision codes
New-style revision codes in use
Using revision codes for board identification

rohan-kapse, 2023-10-23:
hey @aallan could u guide me how can i help, its first time contributing ;)

aallan, 2023-10-23:
> hey @aallan could u guide me how can i help, its first time contributing ;)

Unfortunately, I'm afraid this is not really a first-time contributor sort of issue. I'm going to have to try and figure out how to split the category into sub-topics that are narrative adjacent, which is why I self-assigned.

aallan, 2023-12-31:
Too fast `github-actions` bot, too fast!

aallan, 2024-03-03:
Not yet Gitbot. Not yet.

closed 2024-07-07 by github-actions[bot]