raspberrypi/documentation #3237: Cameras: External Trigger on GS: Please check voltage

njhollinghurst, 2023-10-25:
Another one like #3146 

https://www.raspberrypi.com/documentation/accessories/camera.html#external-trigger-on-the-gs-camera

According to the IMX296 datasheet, XTRIG should be limited to OVDD+0.3V where OVDD=1.8V; but the documentation suggests it's safe to apply 3.3V to XTRIG.

The GPIO->XTRIG connection on the camera board uses a level shifter, but if R11 is removed as instructed then there will be neither level shifter nor pull-up, so it is hard to see how this can safely be driven from a Pico.

Asked on forum here https://forums.raspberrypi.com/viewtopic.php?t=358276#p2148485

njhollinghurst, 2023-10-26:
It's definitely dodgy, the question is what is the most elegant circuit to recommend in the doc. Need to decide whether to recommend a level shifter or if another resistor divider will suffice (maybe with smaller resistances, e.g. 470 or 680).

By the way, Pico can't conveniently do PWM + Open Drain (except using PIO), so an OD/Pull-up style is no good here.

@naushir @6by9 might have comments.

aallan, 2023-10-26:
> [Edit: Not sure why the doc calls for a Pico here rather than a Pi. Just for variety I suppose.]

That's what Simon's summer intern used I think?

simon-martin-rpi, 2023-11-01:
Speaking to Nick. We keep the docs the same, i.e. remove R11. This guy needs to add a resistor divider to his output to reduce 3.3V to 1.8V. I suggest 18K to GND and 15K input resistor.  Since the whole task needs soldering, I guess they will be Ok with it. I hope to respin this board and put a proper connector on it which accepts any voltage at high impedance one day. We have a lot of units with the wrong circuit to sell first though.

njhollinghurst, 2023-11-01:
Got it. So we advise removing R11 in all cases., then can use a divider
Will we need to revise the documentation for the new new board version?

njhollinghurst, 2023-11-01:
Nitpick: The current text is also wrong where it says "This connects GP0 to XTR", since this is GPIO1.

Perhaps the entire paragraph should be something like: "You will need to remove R11, shown in red below, to enable external trigger input. You can then connect an active-low 1.8V logic signal to the XTR pad."

njhollinghurst, 2023-11-10:
I'll attempt a PR for this.

Still wondering about resistor values in #3146 -- it would be paradoxical for us to recommend much larger resistances for a driven signal than for an idle pull-up -- so I might revisit that as well. Will do some testing first!

aallan, 2023-11-10:
> I'll attempt a PR for this.

Thanks @njhollinghurst!

njhollinghurst, 2023-11-13:
> This guy needs to add a resistor divider to his output to reduce 3.3V to 1.8V. I suggest 18K to GND and 15K input resistor.

@simon-martin-rpi Those resistor values seem high compared to the pull-up we suggest [here](https://www.raspberrypi.com/documentation/accessories/camera.html#synchronous-captures). Can I reduce them to (say) 1.8K and 1.5K? That would draw 1mA from the RP2040.

closed 2023-11-16 by aallan