raspberrypi/documentation #3249: Pi5 Kernel build instructions

timonsku, 2023-10-28:
Just noticed the kernel build docs don't have instructions for the Pi5.
Quite a bit changed also in Bookworm with the use of initramfs that also affects older gens, that could probably use updated instructions overall.

migs35323, 2023-11-09:
heh, i tried to follow the docs in a rpi4 with bullseye, and there is no /boot/firmware folder

aallan, 2023-11-09:
> heh, i tried to follow the docs in a rpi4 with bullseye, and there is no /boot/firmware folder

Documentation is written to support our latest operating system Bookworm.

lurch, 2023-11-20:
Partially fixed by #3284

timonsku, 2023-11-20:
Yea I think whats missing is a note to either disable initramfs loading in config.txt or how to generate a new one from the freshly compiled kernel. Would have made a PR but I'm not sure what RPi prefers here

nathan-contino, 2024-02-12:
Not so fast, stalebot!

Vortexxsaman, 2024-04-28:
Mm

fawj2, 2025-06-13:
> Just noticed the kernel build docs don't have instructions for the Pi5. Quite a bit changed also in Bookworm with the use of initramfs that also affects older gens, that could probably use updated instructions overall.

stevedavies-rp, 2026-09-23:
Many thanks for raising this issue. We appreciate your patience.

Instructions for the Raspberry Pi 5 (using KERNEL=kernel_2712 and bcm2712_defconfig) are documented here:

https://www.raspberrypi.com/documentation/computers/linux_kernel.html#natively-build-a-kernel
https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compile-the-kernel

We are now making some minor edits to clarify behaviour around 'instrafms'.

closed 2026-09-23 by stevedavies-rp