raspberrypi/documentation #3252: Update libcamera build instructions

aallan, 2023-10-31:
Need to,

* The "lite" instructions need to replace "sudo pip3 install jinja2"
by "sudo apt install python3-jinja2"

* We need to drop "sudo pip3 install pyyaml ply" for "sudo apt install
python3-yaml python3-ply"

* We can get rid of the "sudo pip3 install --upgrade meson" - hurray!

* The libpisp repository wouldn't let me check it out, I assume the
access permissions need changing!!

* I also notice the meson setup line still includes things like
"-Dpipelines=raspberrypi -Dipas=raspberrypi". I thought we replaced
"raspberrypi" ages ago by "rpi/vc4"?? Anyway, now it wants to be
"-Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp" (or users who
are building for only one of the platforms can obviously omit the
other).

* libcamera-apps were unhappy because it claimed the sensor
configuration was not a pointer (i.e. some confusion about the
presence of std::optional). Maybe that will sort itself out once our
packages all get updated? Possibly we can remove the "sudo ldconfig"
too as that was a workaround for a cmake problem, and we now use
meson/ninja?

aallan, 2023-12-19:
See also #3325.

JimJ92120, 2023-12-20:
hi @aallan 
not sure if relevant to add up here too (i'm also a bit new to Raspberry Pi eco-system)   

following instructions may needs to be more descriptive: https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/camera/rpicam_apps_building.adoc?plain=1#L88   

> `# use -j 2 on Raspberry Pi 3 or earlier devices`   

I've encountered **memory issue** when running the build with `ninja`  on **Pi Zero 2 W** . The process would appear to either get stuck or hang (or very very very slow).  
Over **1h** and it was still stuck on the same step. Retrying many times, with clean install as well didn't make it work.

Device:    Raspberry Pi Zero 2 W  
OS:          Raspberry Pi OS Lite - Bullseye - 32bit version

---

Some research pointed out to possible **RAM limitations**, it would run out of resources when compiling.       
This was also confirmed by running the same compilation on e.g laptop where it builds within 2-3 minutes.  

Using the `-j 2` flag lowers memory usage -  very slow but seems to compile.   
Otherwise, process would hang and never completes (would assume something may crash too at some point).   

----
Suggested to maybe change that **comment** based on **RAM available** vs **device version**?  

I'm not familiar / experienced enough to confirm but would assume, future devices (Pi 6, 7, Pi Zero 3, 4, etc) would encounter similar issue if compiler takes too much resources.

JamesH65, 2023-12-20:
> hi @aallan not sure if relevant to add up here too (i'm also a bit new to Raspberry Pi eco-system)
> 
> following instructions may needs to be more descriptive: https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/camera/rpicam_apps_building.adoc?plain=1#L88
> 
> > `# use -j 2 on Raspberry Pi 3 or earlier devices`
> 
> I've encountered **memory issue** when running the build with `ninja` on **Pi Zero 2 W** . The process would appear to either get stuck or hang (or very very very slow). Over **1h** and it was still stuck on the same step. Retrying many times, with clean install as well didn't make it work.
> 
> Device: Raspberry Pi Zero 2 W OS: Raspberry Pi OS Lite - Bullseye - 32bit version
> 
> Some research pointed out to possible **RAM limitations**, it would run out of resources when compiling. This was also confirmed by running the same compilation on e.g laptop where it builds within 2-3 minutes.
> 
> Using the `-j 2` flag lowers memory usage - very slow but it compiles (30+ minutes). Otherwise, process would hang and never completes (would assume something may crash too at some point).
> 
> Suggested to maybe change that **comment** based on **RAM available** vs **device version**?
> 
> I'm not familiar / experienced enough to confirm but would assume, future devices (Pi 6, 7, Pi Zero 3, 4, etc) would encounter similar issue if compiler takes too much resources.
> 
> It may be possible to build on another (more powerful) machine and ship the build via `ssh` to the Pi - some **symlinks** created during the compilation would cause few issues though.

use -j 1 (or remove the options entirely) on low-memory systems like the Zeros. They will quickly run out of memory when doing parallel builds and then will start swapping, which is very slow in comparison. This will be more noticeable on C++ builds as compiling things like templates is very RAM intensive.

lurch, 2023-12-20:
> OS: Raspberry Pi OS Lite - Bullseye - 32bit version

If you're doing stuff with libcamera then we'd recommend that you use Bookworm rather than Bullseye. And if you're running on a low-memory device like a Zero 2 W then you'll definitely want to stick with the Lite version!

JimJ92120, 2023-12-20:
> If you're doing stuff with libcamera then we'd recommend that you use Bookworm rather than Bullseye. And if you're running on a low-memory device like a Zero 2 W then you'll definitely want to stick with the Lite version!

unfortunately as of now, **Bookworm** isn't available for **Pi Zero 2 W** (v1 too) so not much an option   
**Lite** version for sure

JimJ92120, 2023-12-20:
> use -j 1 (or remove the options entirely) on low-memory systems like the Zeros. They will quickly run out of memory when doing parallel builds and then will start swapping, which is very slow in [...]

good one for pointing out **swap**  

---
with `htop` while compiling, about ~80% on average (clean install, nothing running but that and `htop`):   

`-j 2`:  
[image]  
  
`-j 1`:  
[image]

lurch, 2023-12-20:
> unfortunately as of now, **Bookworm** isn't available for **Pi Zero 2 W**

It **is**, you just need to select "No filtering" in Raspberry Pi Imager, or download it directly from https://www.raspberrypi.com/software/operating-systems/

JimJ92120, 2023-12-20:
forget my comments, i don't know how to read :man_facepalming: 

[image]  

https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/camera/rpicam_apps_building.adoc?plain=1#L13

closed 2024-03-12 by aallan