raspberrypi/documentation #3256: Document which specific release file(s) set new-vs-legacy config.txt boundary

macmpi, 2023-11-04:
Together with `bookworm` release, there's been significant changes in parameters (newer) `config.txt` handles, versus **legacy** one (ref https://github.com/raspberrypi/documentation/commit/e33ec8850cf8b2748722fe503552414c3db2c074).

For those using other distributions than PiOS, but still updating core firmware files (`bootcode.bin`,` *.elf`,` *.dat`), there is **no documented clear-cut** from which file(s) release points / commit hash the boundary is set between new/legacy `config.txt` handling: this makes necessary `config.txt` changes very unclear.

It would really be helpful to have such boundary documented in both `config.txt` documentation pages  (new/legacy) , beyond simple reference to `bookworm` (i.e. a commit hash reference in rpi-firmware repo ?).

@lurch Thanks for consideration.

macmpi, 2023-11-08:
@pelwell would you have any tip on the cliff file/commit?

pelwell, 2023-11-08:
There has been no sudden change to the firmware, except for the introduction of the essentially firmware-less Pi 5 (it's just a basic bootloader). I challenge the idea that there's a cliff in config.txt handling - again, no software change, really - more an increasingly firm nudge for users to give up on the legacy display and camera stacks, with the move to Bookworm marking a clear line of demarcation, a point at which some of the infrastructure around the legacy code can start to be dismantled.

macmpi, 2023-11-08:
interesting...
I was under impression, for instance, "legacy" hdmi-related statements were not working anymore with recent firmware (`hdmi_safe=1`, etc): will test again.

Also surprising current `config.txt` doc does not talk much about `vc4-kms-v3d`-related stuff, although it is explicit requirement in bookworm release `config.txt`.

> a point at which some of the infrastructure around the legacy code can start to be dismantled.

Would this mean things like `vc4-kms-v3d` will sone become the non-explicit default and won't need to be mentioned in `config.txt `anymore (a bit like default change for `krnbt`)?

pelwell, 2023-11-08:
The use of the kms overlays disables the in-firmware display driver.

> Would this mean things like vc4-kms-v3d will sone become the non-explicit default and won't need to be mentioned in config.txt anymore (a bit like default change for krnbt)?

That is the logical endpoint. It's already the case with the upstream kernel.

folknor, 2024-01-15:
I have an RPi2 and RPi3 hooked up to TVs at home, and when I updated those some time ago the HDMI output started behaving really weirdly, so I just rolled them back after trying to tweak config.txt a few times.

Now, I bought an RPi4 for a TV at work, and the TV just flickers on and off constantly. The more throughput is required by the connection, the more it turns on and off, so reducing the resolution and/or refresh rate both reduce the "flickering" (which is the wrong word, because the signal is dropping so the TV turns black.) At 1920x1080 60hz it just constantly does it, more than one time per second.

So what do you think I did? I tried all the config.txt settings I learned to try for RPi3.

The docs are horrendous for this issue.

Also, if I were actually looking for docs for the various hdmi config tweaks for RPi3, where would I find those now?

lurch, 2024-01-15:
> Now, I bought an RPi4 for a TV at work, and the TV just flickers on and off constantly. The more throughput is required by the connection, the more it turns on and off

It's probably worth creating a separate issue for that at https://github.com/raspberrypi/linux/issues rather than trying to discuss the problem in the documentation repo.

> I have an RPi2 and RPi3 hooked up to TVs at home, and when I updated those some time ago

Note that for Bookworm we're not supporting in-place upgrades, see https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

> Also, if I were actually looking for docs for the various hdmi config tweaks for RPi3, where would I find those now?

As long as you're using a fresh install of Bookworm, see https://www.raspberrypi.com/documentation/computers/config_txt.html#video-options

folknor, 2024-01-15:
> > Also, if I were actually looking for docs for the various hdmi config tweaks for RPi3, where would I find those now?
> 
> As long as you're using a fresh install of Bookworm, see https://www.raspberrypi.com/documentation/computers/config_txt.html#video-options

Right, which doesn't link at all to https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#hdmi-mode

And the explanation at the top there that says `The config.txt options described here are considered legacy settings, are not used by Raspberry Pi OS Bookworm` (https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#legacy-options) is not presented in the inverse anywhere in the new documentation.

I'm not running bookworm, and finding these docs for the old software and RPi4 was extremely difficult.

That's what my comment was about - I didn't comment here to get help with anything specific. I commented to point out how difficult it is to find the relevant information.

aallan, 2024-03-17:
Too soon Github bot, too soon.

stevedavies-rp, 2026-09-23:
Will not document legacy settings.

closed 2026-09-23 by stevedavies-rp