raspberrypi/documentation #3258: pico-sdk/hardware.html freezes the tab because of a JS busy loop

mordae, 2023-11-05:
Hi,

it seems that [scrollToElement()](https://github.com/raspberrypi/documentation/blob/803c08d739656dd6bad3be2a4afcee910c4467c8/jekyll-assets/_includes/scripts.html#L45) is somewhat broken.

As you can see in the following screenshot, it makes **thousands** of attempts to locate the element in this very long page before giving up. Obviously the whole time, 38 seconds for me, the tab is frozen and Firefox even advises interrupting the script. It happens to me both in Firefox and Chromium on Fedora 39 (beta).

[image]

aallan, 2023-11-06:
@nelliemckesson Can you take a look? Thanks!

nelliemckesson, 2023-12-08:
Confirming issue also occurs in Chrome: 

[image]

PR #3315 (2023-12-10): Fix scrollToElement infinite loop
fixes #3258
closed 2023-12-10 by aallan