raspberrypi/documentation #3266: Add information on creating Read-Only filesystems

aallan, 2023-11-08:
How read only filesystems work has changed in Bookworm. See raspberrypi/bookworm-feedback#137 for details. More details need adding to the documentation. There is already something in raspi-config about the overlay filesystem,

![image](https://github.com/raspberrypi/bookworm-feedback/assets/53625/c8d87447-3fb0-4248-b1fd-3c70ea46900

but this option doesn't necessarily do what some end users would expect. Should add more information to [this section](https://www.raspberrypi.com/documentation/computers/configuration.html#overlay-file-system) I guess?

lurch, 2023-11-09:
It's not only in raspi-config, it's also in the Raspberry Pi Configuration GUI tool.

aallan, 2024-03-17:
Too soon Github bot, too soon.

theepicflyer, 2024-05-15:
Agree! I just spent an afternoon figuring out how to set overlayroot to overlay just root and not /home. I have a separate /home partition and want to keep that writable for user data.

Mainly I was misled by this project on github also called [overlayroot](https://github.com/chesty/overlayroot). I thought this was overlayroot and tried to customize it this way. The Debian [cloud-initramfs-tools](https://salsa.debian.org/cloud-team/cloud-initramfs-tools) does not come up in a Google search, and I only found it after I went through the Debian package browser.

Please add documentation to customize overlayroot!

nathan-contino, 2024-07-15:
Still important. Just need more resources to get to it.