raspberrypi/documentation #3275: Error installing library for Build HAT

Gun-neR, 2023-11-14:
New install of Legacy OS on RPi4
Ran all the apt-get updates
Unable to install library for Build Hat (LEGO) using directions in documentation...

```

Installing the Python Library
Install the Build HAT Python library. Open a Terminal window and type,

$ sudo apt install python3-build-hat

```
pi@piLEGOHAT:~ $ sudo apt install python3-build-hatpip3
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-build-hatpip3

Gun-neR, 2023-11-14:
A little more Googling and found this... and it seems to work.  Says it is Installed, but have yet to test.  Meanwhile, documentation might need updating??

sudo pip3 install buildhat

aallan, 2023-11-14:
@ghollingworth Is the build hat python library apt package only available for Bookworm currently?

aallan, 2023-11-14:
> Meanwhile, documentation might need updating??

The documentation is correct for Bookworm, the current distribution of Raspberry Pi OS.

Gun-neR, 2023-11-14:
> > Meanwhile, documentation might need updating??
> 
> The documentation is correct for Bookworm, the current distribution of Raspberry Pi OS.

Doesn't mention that fact, just says compatible with all 40 pin GPIO based RPis.  So yet another thing that might need to be added to the docs.  Too many remaining issues with Bookworm and legacy codes, etc.... IE, I couldn't even get connected to my headless setup with VNC so I went back to legacy and it just worked.

Doesn't matter much anymore to me.  My issue and solution is at least now searchable... I couldn't find anything specifically related to my error msg, but only stumbled upon the other/older? method of install thanks to some GoogleFoo :)

aallan, 2023-11-14:
Reopening issue. Either we need a NOTE block for legacy in the docs to say to use pip, or we need an apt package for the library for Bullseye.

Gun-neR, 2023-11-14:
No problem... wasn't sure who should close it, and I did kind of open it a bit prematurely (should have trusted my Googlefoo more , hah! ) but I had left it for the next person who, like me, couldn't find a solution in the relevant locations.  

It is now in your hands.  Thanks!

aallan, 2024-03-17:
Too soon Github bot, too soon.

closed 2024-06-06 by nathan-contino