raspberrypi/documentation #3285: Documentation of available RP1 GPCLK features

pavhofman, 2023-11-20:
Based on https://forums.raspberrypi.com/viewtopic.php?p=2158890#p2159063 , I would like to ask for exposing information on the GPCLK feature shortly mentioned in the existing RP1 docs draft, and partly introduced in clk_rp1.c, especially the clk <-> GPIO matrix available for general use.

Specifically I am interested in MCLK GPCLK input/output/output tap for I2S which is important for connecting external ADC/DAC chips with MCLK input pin (i.e. all ADC/DAC chips without an internal BCLK -> MCLK PLL).

pavhofman, 2024-01-14:
Hi please any chances? The board is already available, vendors could start building new hats and the MCLK signal for 8ch DAC/ADC would be very handy, especially considering the dedicated audio PLL can be integer multiple of  standard audio frequencies, avoiding jittery fractional division. Thank a lot in advance!

aallan, 2024-01-16:
Unfortunately, it's unlikely we'll be adding further detail to the [draft RP1 Peripherals](https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf) book in the near to medium term.

P33M, 2024-01-17:
@aallan the clock infrastructure is a user-facing feature, and has been explicitly designed with the intent of supporting a wide range of use-cases. 

This issue should not be closed.

aallan, 2024-01-17:
> This issue should not be closed.

Who should I be looking at to write the requested documentation?

P33M, 2024-01-17:
Me.

But I need to 
a) find time to do so
b) refresh my knowledge of what non-functional (test, debug, ATE) clocks were added to the already enormous functional clock table that need to be omitted.

introspektor, 2024-01-28:
Well thats disappointing. I bet (I know) there are a number of developers that rushed out to buy some RP5 based on the pre-release documentation and pinout for multichannel audio. this feature was sorely lacking in previous rpi hardware and I know we breathed a sigh of relief and excitement when reading the RP1 document and rushed out to buy some boards for development immediately. To have the hardware to complete the task in hand, knowing it has the capability and be left waiting is a bit frustrating. I appreciate things take time and we are all busy, so dont take me the wrong way. 

I shall wait with baited breath.

cleverca22, 2024-02-16:
another question ive got, is if the PIO block can be fed by an external clock, or only internal clock sources

P33M, 2024-02-16:
PIO's only clock source is the system clock, i.e. the same as the M3s.

cleverca22, 2024-02-16:
how can the system clock divider be changed? can it source from an external input like the rp2040?
what else is running on the system clock, other then the stuff that is clearly a child in linux's clk_summary

petercsmith-x, 2025-06-27:
Are there any updates on this? I've been stuck trying to get an I2S MCLK clock signal out on GPCLK0 and I can't find documentation. Much appreciated.