raspberrypi/documentation #3292: alsactl restore commands for RPi Codec Zero on Raspberry Pi 4 (Bookworm) doesn't work

robespo72, 2023-11-21:
I've the follow setting

Raspberry Pi 4
RPi Imager 1.7.5
Debian Bookworm 10/10/2023
debian version 12.1

IQaudIO Codec Zero (Black edition, pre-rebranding)
Copyright IQaudIO Limited 2019 (PCB 020/v1)

grep -a . /proc/device-tree/hat/*
/proc/device-tree/hat/name:hat
/proc/device-tree/hat/product:Pi-CodecZero
/proc/device-tree/hat/product_id:0x0013
/proc/device-tree/hat/product_ver:0x0001
/proc/device-tree/hat/uuid:e15c739c-877d-4e29-ab36-4dc73c21127c
/proc/device-tree/hat/vendor:IQaudIO Limited www.iqaudio.com

lsmod | egrep 'da7213|iqaudio'
snd_soc_da7213         65536   1
regmap_i2c             16384   1 snd_soc_da7213
snd_soc_iqaudio_codec  20480   0
snd_soc_core           274432  3 snd_soc_iqaudio_codec,snd_soc_bcm2835_i2s,snd_soc_da7213
snd_pcm                139264  6 snd_soc_bcm2835_i2s,snd_bcm2835,snd_compress,snd_soc_da7213,snd_soc_core,snd_pcm_dmaengine

I slavishly followed the official guide:
https://www.raspberrypi.com/documentation/accessories/audio.html

who states that when I meet "IQaudIO Limited www.iqaudio.com" you have to modify the file /boot/config.txt as:
#dtparam=audio=on
dtoverlay=
dtoverlay=rpi-codeczero

I did it.

I created the file .asoundrc ...
pcm.!default {
        type hw
        card Zero
}

I restarted Raspberry for the first time, placed it in shutdown again, applied the "IQaudio Codec Zero" (black edition) HAT while Rasp was in shutdown and restarted it again. I have downloaded the "state" ...

git clone https://github.com/raspberrypi/Pi-Codec.git

and runned the command ...

sudo alsactl restore -f /home/pi/Pi-Codec/Codec_Zero_OnboardMIC_record_and_SPK_playback.state
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
alsactl: set_control:1475: Cannot write control '2:0:0:Mic 1 Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:Mic 2 Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:Aux Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:Mixin PGA Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:ADC Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:DAC Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:Headphone Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:DAC EQ Switch:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:DAC EQ1 Volume:0' : Remote I/O error
.
.
.
.
.
.
alsactl: set_control:1475: Cannot write control '2:0:0:ALC Max Threshold Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:ALC Max Attenuation Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:ALC Max Gain Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:ALC Min Analog Gain Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:ALC Max Analog Gain Volume:0' : Remote I/O error
alsactl: set_control:1475: Cannot write control '2:0:0:ALC Anticlip Mode Switch:0' : Remote I/O error

I started to look for info and it seems that the lsmod output shown above indicates that the sound card drivers have been correctly installed, so I thought that the IQaudIO Codec Zero firmware needed to be updated, so I performed this sequence as the guide indicates:

sudo apt update
sudo apt install rpi-audio-utils
sudo rpi-audio-flash

the last of these three commands, on the second screen, returned this message:
"Detected a programmed Raspeberry Pi Codec Zero pHAT"

Does this mean that the EEPROM is already updated?
However it did not allow me to reprogram the EEPROM!

Furthermore in my case it seems that there was nothing to be joined to eliminate the write protection of the EEPROM as I am in the following condition:

"In some cases the two pads may already have a 0 ohm resistor fitted to bridge the write-protect link, as illustrated in the picture of the Codec Zero board above"

Then I restarted my Rasp and ran again the command:
sudo alsactl restore -f /home/pi/Pi-Codec/Codec_Zero_OnboardMIC_record_and_SPK_playback.state

Nothing to do. Nothing has changed.

I followed issues #2891, #2894, #2961 to look for useful information for me, but without success
Can anyone help me? What is wrong?

aallan, 2023-11-21:
I've asked internally to see if we can get someone to figure out what's going on here.

robespo72, 2023-11-21:
Thank you @aallan!!

robespo72, 2023-11-30:
Ok @aallan, I'll try to do it. Thanks!

robespo72, 2023-12-05:
hello to everyone!
I followed @Lurch ' suggestions here... https://github.com/raspberrypi/bookworm-feedback/issues/134#issuecomment-1831627444.
It's frustrating, it doesn't work. Always fail the "alsactl restore" step and "alsamixer" sees the mixer voices, but does not allow you to adjust the levels.
Using "aplay" you can't hear any sound, probably because the levels are at 0 and not changeable

lurch, 2023-12-05:
Can you please copy'n'paste (or upload a photo) of the exact error messages you're getting? As @pelwell notes in https://github.com/raspberrypi/bookworm-feedback/issues/24 the "error" you're seeing might be "just an ugly success" (i.e. a harmless warning).
Which exact state-file did you use, and which volume levels in alsamixer did you try increasing?

robespo72, 2023-12-14:
Hi @lurch, I repeated everything. The sound card is the one you see in the picture IQaudIO_Codec_Zero_on_RP4.jpg, the one referred to as "IQaudIO Codec Zero" (Copyright IQaudIO Limited 2019 - PCB 020/v1), therefore the black one and not the green one. Mine is a Rasp Pi 4, and the OS version is "Debian Bookworm 10/10/2023". The "state" file I used is this

Codec_Zero_OnboardMIC_record_and_SPK_playback.state

downloaded with the command suggested by the official guide (https://www.raspberrypi.com/documentation/accessories/audio.html):
git clone https://github.com/raspberrypi/Pi-Codec.git

The intention was to use the onboard MIC (MEMS Mic2) for recording and an 8ohm & 0.4W speaker connected to the screw terminals as a playback (see picture 8ohm_0.4W_speaker.jpg). As the official guide suggests I launched the command:

sudo alsactl restore -f /home/pi/Pi-Codec/Codec_Zero_OnboardMIC_record_and_SPK_playback.state

the errors are still the same as those I reported in the first post of 21st November, nothing has changed.

The alsamixer command allows me to view the menu, but all the levels are stuck at volume zero and for none of them it is possible to increase

[image]
[image]

lurch, 2023-12-15:
Would you like to give https://github.com/raspberrypi/Pi-Codec/pull/8 a test and see if that works any better for you?

robespo72, 2023-12-18:
Could you have a look to this short video please?
Thank you in advance!

https://github.com/raspberrypi/documentation/assets/151676414/6c791055-dae0-48e9-a315-667c1f0f23d5

lurch, 2023-12-18:
Hmm, I don't see those "Remote I/O Error" messages on my card.
[image]

Any idea what might be causing those @pelwell ?

pelwell, 2023-12-18:
In the world of I2C, Remote I/O error (`REMOTEIO`) means "I couldn't talk to the I2C device", which is usually a sign of a hardware problem.

aallan, 2023-12-18:
@robespo72 Can you drop an email to applications@raspberrypi.com

robespo72, 2024-01-10:
Ok, I give up, I think the same .. maybe there are some hardware problems. I suppose we can close this issue.
Thank you to everyone!!

closed 2024-01-10 by aallan

viper9503, 2024-01-26:
> Hello, i am facing some errors but not the REMOTEIO ones, i am so lost i have been trying to fix this issue since the past week and there is not way to fix it. i keep getting this error

alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2