raspberrypi/documentation #3308: Revisit camera sync and external trigger

njhollinghurst, 2023-12-07:
Another installment in the #3146 #3237 saga.

It seems BB's original circuits were designed to overcome protection/clamping diodes or back-power the 1.8V supplies of cameras that were powered down! They were out of specification and can't be reverted to. But my recent "fixes" suffer voltage drops at startup, introducing a risk of spurious triggering.

Specifically,
- The 10K:10K divider pull-up is not strong enough to give a high level for more than one HQ camera, when some cameras are powered down.
- Some readers have interpreted that 10K resistance as normative for an external driver (though it was indicated for a pull-up).
- The voltage divider for Pico GPIO ->GS Camera XTR now gives safe voltage levels, but its quiescent state (when Pico is not driving the GPIO) will be low instead of floating.

The main part of the fix is a kernel/DT change (currently in the works [here](https://github.com/raspberrypi/linux/pull/5774)) to allow the camera's 1.8V supply rail to stay up even when the camera is not streaming, so it won't clamp its inputs to 0V.

Then we'll need to:
- Describe how to use the new overlay parameter to keep the cameras powered on;
- Remove the external fixed pull-up circuit for HQ Cameras;
- Figure out the best startup order for Pico -> GS Camera;
- Document the configuration file used to control CSI-2 timeouts (since it's needed but was undocumented);
- Maybe fill in the fourth combination: How to drive multiple HQ cameras from an external trigger (since people seem to want to do it).

I've opened this as an issue rather than a PR to give me time to think about it and to invite comments.

lurch, 2023-12-07:
Pinging @tobybroberts as this will affect one of the projects that he's working on.

kyrikakis, 2024-04-20:
Hi there,

Do you plan to add instructions for boards with dual CSI camera interfaces like CM4 and Pi5? I have seen there is a commit here allowing to explicitly configure each camera to it's respective mode: https://github.com/raspberrypi/linux/commit/5b432b8de1ce420091aad42d2cadc0125c643038#diff-d10f1f1ae2309b3d1d2f5f2f7fc69598753558dc7e81215371f38b23a235cde2

Thanks,
Stefanos

lurch, 2024-04-23:
:point_up:  ping @davidplowman and @naushir

lurch, 2025-05-21:
@njhollinghurst Can this be closed now, in light of #4089 ?

njhollinghurst, 2025-05-22:
Ah, I'd forgotten about this one. From the original shopping list:

- Figure out the best startup order for Pico -> GS Camera;
- Maybe fill in the fourth combination: How to drive multiple HQ cameras from an external trigger

are outstanding, but those are secondary. In general this stuff is being obsoleted by software sync. So yes I think it can be closed.

closed 2025-05-22 by njhollinghurst

njhollinghurst, 2025-05-22:
@kyrikakis Yes, the properties for individual IMX477 cameras are now documented (though I didn't mention the use of 2-port or 4-port camera mux).