raspberrypi/documentation #3327: Harvesting the forums

aallan, 2023-12-20:
We're probably at the point where we should try and move some of the stuff that is only on the forums into the main documentation, e.g. This troubleshooting thread around booting, https://forums.raspberrypi.com/viewtopic.php?t=58151. This duplicates a lot of the documentation, but does have some things the documentation doesn't cover.

aallan, 2023-12-20:
Unsure at this point whether we should just grab content and get it inserted in the right places. Or make a list of possible content topics, and grab the relevant content after we have a structure for the new site. Will think about it over Christmas.🎄🎁

LizUpton, 2024-04-28:
Not so fast, bot. Stale label removed. Nate, we should talk about this next week.

nathan-contino, 2024-07-01:
Still on the radar.

closed 2024-11-27 by nathan-contino