raspberrypi/documentation #3350: pi5: peripheral address missing, and bcm_host_get_peripheral_address() broken

cleverca22, 2024-01-06:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#peripheral-addresses
this page makes no mention of the peripheral base addr on the pi5

`bcm_host_get_peripheral_address()` also claims the peripherals start at just `10`

```c
#include <stdint.h>
#include <sys/mman.h>
#include <stdio.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>
#include <stdlib.h>
#include <bcm_host.h>

int main(int argc, char **argv) {
  uint64_t physaddr;
  physaddr = bcm_host_get_peripheral_address();
  printf("addr %lx\n", physaddr);
  return 0;
}
```
this code returns `addr 10`

aallan, 2024-01-06:
On Pi 5 GPIO is controlled via [the RP1](https://www.raspberrypi.com/documentation/microcontrollers/rp1.html). See the RP1 Peripherals document, https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf.

cleverca22, 2024-01-07:
@aallan i know most gpio is controlled by the RP1, and in this use-case, i didnt want an RP1 register
i was after the old ST_CLO register in the VPU timer, which i did find after some brute-forcing

but that does raise an interesting problem
with how the pi5 has re-arranged all of the SoC registers, and moved most stuff the user wants to the RP1, the old `bcm_host_get_peripheral_address()` is kind of useless
any offsets you would have done from there are now going to be wrong, no matter which base it returns

aallan, 2024-01-07:
@pelwell thoughts here?

pelwell, 2024-01-07:
The source code for the bcm_host library lived in the userland repo, the README for which states:

> This repo is ancient and deprecated.
> 
> It largely contains code using proprietary APIs to interface to the VideoCore firmware. We have since move to standard linux APIs.
> 
> V4L2, DRM/KMS and Mesa are the APIs you should be using.
> 
> The few useful tools from here (dtoverlay, dtmerge, vcmailbox, vcgencmd) have been moved to the raspberrypi/utils repo.
> 
> Code from here is no longer installed on latest RPiOS Bookworm images.
> 
> If you are using code from here you should rethink your solution.
> 
> Consider this repo closed.

There is no correct return value for `bcm_host_get_peripheral_address()` on Pi 5 other than NULL, but updating it would imply that the library still has a life.

aallan, 2024-01-08:
@pelwell At that point I'm presuming this section https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#peripheral-addresses needs rewriting?

pelwell, 2024-01-08:
I think we should just delete it. What do you think, @popcornmix ?

popcornmix, 2024-01-09:
I agree. The repo the code is from has been closed, and I don't think the code is useful enough to resurrect it
(especially as most uses are unlikely to be applicable to Pi5 which uses RP1 for many equivalent peripherals).

aallan, 2024-01-09:
Okay, I'll yank that entire section.

aallan, 2024-01-15:
Removed entire section in ddfba6d. Will go to the live site next time I push to production.

closed 2024-01-15 by aallan

lurch, 2025-07-07:
Hello @Seneral , I've discussed this with some of the other engineers, and they tell me:
* bcm_host was removed because large parts of it no longer make sense in a Pi 5 world
* For getting raw peripheral addresses, you should be reading the values from DeviceTree (although we'd always advise you to access peripherals via their standard Linux APIs where possible, rather than directly banging the registers, as this isolates you from future hardware changes)
* Direct QPU access is no longer possible on Pi 5 (access isn't implemented in the Pi 5 EEPROM)
* For Pi 4 and earlier, direct QPU access is now deprecated (because the OpenGL driver is now implemented on the ARM side rather than in the firmware), but still possible - see https://github.com/raspberrypi/linux/issues/6660 for some example code