raspberrypi/documentation #3351: 3B+ revision a020d4 is missing.

stephan57160, 2024-01-08:
Just received a few 3B+ with some unknown revision:
```
$> grep -e Model -e Revision /proc/cpuinfo
Revision	: a020d4
Model		: Raspberry Pi 3 Model B Plus Rev 1.4
$> 
```

Still digging in documentation, but I can't find this one under: 
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/revision-codes.adoc

aallan, 2024-01-08:
Ping @JamesH65 who probably is the best person for this one?

JamesH65, 2024-01-08:
That's just the v1.4 board (New PMIC), see PCN here https://pip.raspberrypi.com/categories/797-pcn/documents/RP-003337-PC/Pi3B-Revision-9-PCN.pdf. I'll update the page when I get the chance. Note that the table in the docs is not guaranteed to cover everything, and should not be relied upon for board identification - use the scheme noted later on the page.

JamesH65, 2024-01-09:
See https://github.com/raspberrypi/documentation/pull/3354

closed 2024-01-09 by aallan