raspberrypi/documentation #3353: Rotating 7-inch Touch Display is a mess -- needs proper research then update of config-tools and then new instructions in doc

jdg71nl, 2024-01-09:
(this is my "first create-issue", so please help if forget something)

I am using the **7-inch Touch Display accessory** for many years now, for both business and personal projects, and I am always fighting with the "change-orientation" settings.

The two most popular display housing options (Multicomp and Pibow) both "force" you to install the display upside-down, relative to the housing if sitting on a flat surface, or desk.
The best solution would be for the housing to allow hardware installation/mounting in 2 positions (0 and 180 degrees).
But that is not the case, so we are stuck with **software-rotation**.

I have seen and used the rotation settings in the official website:
https://www.raspberrypi.com/documentation/accessories/display.html
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/display/display_intro.adoc

But I always end-up with either the text/console being 180-rotated, and the GUI not.
Or reverse.
And then there is the issue of the touch-interface rotation being independent from the display-rotation.
And finally, the VNC-server does not "follow" the display-rotation.

I realize that fixing this issue, is foremost the addition/update of the driver config (in cmdline.txt or config.txt), more that just an update of this doc.
But how do I file an github-issue for the correct "7-inch Touch Display" driver (-config) ?

This is a screen-recording of VNC with the pointer being reversed:
<d231229-JDG-Screen-Recording--VNC-pointer-reversed.mp4>
https://github.com/raspberrypi/documentation/assets/12169650/3e2176b6-7f6d-4b8d-b2cf-1f59b1a9cb3c

6by9, 2024-01-09:
Setting `video=DSI-1:800x480@60,rotate=180` in cmdline.txt will invert the console by setting rotation on the primary plane for that output. However currently Wayfire doesn't reset that option, so you end up with the background plane rotated when the cursor plane isn't.
There is a one line fix for that to a package called wlroots that will resolve that. I've created the [patch in the mainline project](https://gitlab.freedesktop.org/wlroots/wlroots/-/merge_requests/4508), and that should ripple through to Raspberry Pi OS fairly quickly. With that you should be able to use both the cmdline.txt setting and the Screen Configuration app in Wayfire and get consistent results.

Wayfire will want the touch matching the default orientation of the display itself. When assigned to a display, transforms on the display will also be reflected in the touch input as well. I believe you can insert a custom transform in libinput/Wayfire, but it just becomes more complex.
Inverting the touch via device tree is going to cause grief against Wayfire. When in the console there is no standard user of touch events, so largely touch there is redundant. If using a library to take touch events, then generally those have a mechanism to handle inverting to match the display output, and that should really be investigated instead of changing DT.

WayVNC reversing the mouse pointer looks to be a bug in (probably) WayVNC, as I've reproduced that using TigerVNC as the viewer. It's for any rotated display not just the DSI panel, so implies that something is passing around co-ordinates/pointer movements without having checked for transforms first.
There is a [wayvnc issue](https://github.com/any1/wayvnc/issues/25) that appears identical from 2020, but claims to have been resolved. That was against sway and wlroots, so potentially it's reoccurred in Wayfire.

aallan, 2024-01-11:
@nathan-contino Can you insert the core of the above 👆 into the documentation on the 7-inch Display!

6by9, 2024-01-11:
I will say that we are looking at whether there is a more uniform way of configuring rotation over the various display options, but it's not a totally straight-forward task.

nathan-contino, 2024-01-11:
@aallan The `cmdline.txt` is already in the doc. Do we have any other concrete solutions for this problem besides waiting for Wayfire and waynv to fix the issue, and waiting for the wlroots fix to percolate down to Raspberry Pi OS?

I prefer not to document known bugs, but this issue seems persistent enough that we might need to add something to the docs about debugging and fixing it. Is there a known and working temporary solution to invert input via device tree, or (preferably) invert input via touch input library?

I would like to add more information to the docs about this, but I'd like to avoid adding a paragraph without an actionable solution. Even if it's not necessarily the cleanest or best solution right now.

aallan, 2024-01-11:
Sounds like a question for @6by9 if the `cmdline.txt` stuff has already made its way into the documentation, is there anywhere we should be pointing folks with questions around the 7-inch Display and rotation? Github issue, Forum thread perhaps? At minimum, it'd be good to add a NOTE block which says "There is a know bug where foobar, for more information see xyzzy."

6by9, 2024-01-11:
We have the Official Display subforum at https://forums.raspberrypi.com/viewforum.php?f=108
The sticky FAQ there is hideously out of date, but a new sticky would probably be the most sensible thing to cover known issues.
Otherwise the bookworm-feedback repo generally catches other issues.

`cmdline.txt` is in there, but recommends using the dtoverlay `invx` and `inxy` overrides to flip the display. That ought to be caveated on not involving any window manager, and ideally to use whatever input transformation facilities there are in the chosen input library rather than inverting at the driver level.
There's also no clear separation between "this applies for GUIs" vs "this is console only".

JamesH65, 2024-01-11:
If you can write a new sticky, I'll remove the old one (presuming there is nothing left in there of interest)

videoman, 2024-02-23:
This is still an issue with Raspberry pi and WayVNC
/etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

dpkg -l of wayvnc package.
ii  wayvnc         0.8.0-rc0-2  arm64        VNC server for wlroots-based Wayland compositors

/boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=81e5547f-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles video=HDMI-A-1:1280x400@60234,rotate=90

~/.config/wayfire.ini
[output:HDMI-A-1]
mode = 400x1280@60234
position = 0,0
transform = 90

Switching back to X11 server and vncserver (RealVNC) fixes the issue. I believe that this is a wayvnc issue. All local USB and bluetooth mouse movements work as expected on the console/screen. Touch screen also registers correctly. It's just the VNC session that has reversed X and Y mappings. Before I upgrade the wayvnc service, upon connect it would show the orientation off by 90 degrees before it would do a screen refresh. Once the screen refreshed, it would then show the correct orientation. After wayvnc upgrade to 0.8.0-rc0-2 upon connection it would now show the right screen orientation upon connect. Still has reversed X and Y mouse movements. 

I was hoping to find documentation in wayvnc on the config file to reverse the mouse movements, but did not see anything in the config documentation for that.

lurch, 2024-02-24:
:point_up:  ping @any1

steveiliop56, 2024-03-12:
@6by9 any updates on the pr? Is there any temp fix for that?

6by9, 2024-03-12:
> @6by9 any updates on the pr? Is there any temp fix for that?

What exactly are you referencing by "that"? There are a number of things being discussed in this thread.

aallan, 2024-03-12:
Probably some overlap with #3184.

steveiliop56, 2024-03-12:
> > @6by9 any updates on the pr? Is there any temp fix for that?
> 
> What exactly are you referencing by "that"? There are a number of things being discussed in this thread.

The inverted mouse on vnc.

6by9, 2024-03-12:
> > > @6by9 any updates on the pr? Is there any temp fix for that?
> > 
> > 
> > What exactly are you referencing by "that"? There are a number of things being discussed in this thread.
> 
> The inverted mouse on vnc.

Not one I've invested significant time in, beyond replicating.
Contrary to my previous comment, it looks to be in Wayfire as using alternate compositors (sway or labwc) doesn't show the issue even though they are both still using wlroots and wayvnc.

@cillian64 is it worth raising it with the Wayfire devs? I think we may have mentioned trying to replicate on x86_64 first (need to find some hardware to do that).

steveiliop56, 2024-03-12:
I saw that you made a pr to wlroots but it didn't actually get merged. It's on a review state and they left you a comment on fixing something.

6by9, 2024-03-12:
> I saw that you made a pr to wlroots but it didn't actually get merged. It's on a review state and they left you a comment on fixing something.

Unrelated to the vnc issue.
That was if you had added `video=....,rotate=180` or similar to rotate the console, but would leave the primary plane inverted but not the cursor plane.

steveiliop56, 2024-03-12:
Oh alright so that's not the problem. Ok

lurch, 2024-04-17:
@nathan-contino See also the notes in https://github.com/raspberrypi/bookworm-feedback/issues/161 (but I _think_ they might be related to 3rd-party screens, and not our offiical touch screen? )

nathan-contino, 2024-06-06:
This _should_ be fixed now, in VNC, desktop, and console. If anyone is still experiencing problems, please file a separate issue -- this one has sprawled a bit.

closed 2024-06-06 by nathan-contino

steveiliop56, 2024-06-06:
@nathan-contino by fix you mean something in the pi os? Wayfire fix? Or just because pi os is switching to labwc?

nathan-contino, 2024-06-06:
My understanding is wayfire and wayvnc fixes.

steveiliop56, 2024-06-06:
Oh wow if wayfire fixed it (it was a wayfire bug) that's a big deal! I will switch back to wayfire to test.

steveiliop56, 2024-06-06:
Doesn't seem to work in my testing.

steveiliop56, 2024-06-06:
I am rotating the screen to the right and mouse inverted rotating back to normal mouse fine.

nathan-contino, 2024-06-06:
Rats. Thanks for testing. I'm going to leave this documentation issue closed because AFAIK there's nothing to document -- this is a wayfire bug, and the fix is reliant on wayfire's developers fixing it. Happy to add a note with a workaround if you have one working!

steveiliop56, 2024-06-06:
Yeah the workaound is simple just switch to labwc. Firstly install the package `sudo apt install labwc` then using `raspi-config` go to Advanced Options select Wayland then select labwc reboot and issue fixed. Just note that labwc a) is currently in beta b) it will probably reset some options like keyboard layout.

nathan-contino, 2024-06-06:
Thanks. I'm adding this as a workaround in the docs until someone gets to the bottom of the Wayfire issue. https://github.com/raspberrypi/documentation/pull/3705