raspberrypi/documentation #3479: Kill the Pi 5 page with 🔥

aallan, 2024-01-19:
Introduced because we wanted somewhere pre-launch to put information about Raspberry Pi 5, it's time [this section of the documentation](https://www.raspberrypi.com/documentation/computers/raspberry-pi-5.html) was broken up and distributed into the correct places in the docs — or new places created if no such places exist.

Then the Pi 5 section needs to be deleted.

LizUpton, 2024-01-19:
https://github.com/raspberrypi/documentation/edit/develop/documentation/asciidoc/computers/raspberry-pi/power-supplies.adoc is one SCHPLENDID example of a place where the relevant content has ended up on the Pi 5 page.

aallan, 2024-01-19:
> https://github.com/raspberrypi/documentation/edit/develop/documentation/asciidoc/computers/raspberry-pi/power-supplies.adoc is one SCHPLENDID example of a place where the relevant content has ended up on the Pi 5 page.

Related to #3490 but not the entirety of that issue — everything else in that section also needs updated.

aallan, 2024-01-19:
See also #3485

closed 2024-02-19 by aallan