raspberrypi/documentation #3486: Add info on GPIO drive strength for Pi 5

LizUpton, 2024-01-19:
https://github.com/raspberrypi/documentation/edit/develop/documentation/asciidoc/computers/raspberry-pi/gpio-pad-controls.adoc

aallan, 2024-03-25:
No so fast github bot.

njhollinghurst, 2025-11-07:
I just came to give this a bump, and in particular to ask that the "safe current" limits (per-pin and total) be clearly specified for the Pi 5 series (as they are for Pi 0-3).

It needs clarification for Pi 4 series too (which currently has a note to say that drive strengths are halved, but it doesn't explicitly say if the safe limits are different?)

njhollinghurst, 2025-11-07:
CM4 and CM5 datasheets both give 50mA for the total source/sink limits, but neither specifies a per-pin safe limit.