raspberrypi/documentation #3505: Explanation of LED behaviour on different models of Pi

lurch, 2024-01-22:
On some Pis (e.g. Raspberry Pi 2, Raspberry Pi 4) there are separate red and green LEDs, and (AFAIK) the red LED is normally-on to indicate that 5V power is being supplied to the Pi (I think on some models the red LED also flashes off to indicate under-voltage warnings?), and the green LED is normally-off but flashes on to indicate SD-card activity.
On some Pis (e.g. the Raspberry Pi Zero and Zero 2 W) there's only a single green LED, which is normally-on but flashes off to indicate SD-card activity.
And on the Raspberry Pi 5 there's a single bi-colour LED, which is normally-on green but flashes off to indicate SD-card activity, and turns red to indicate that the Pi is in the "shutdown" state (and can be woken up by pressing the power button).

With all these variations, _perhaps_ it's worth adding a section to the documentation explaining all of this? (and of course my descriptions above should be double-checked as they may not be entirely accurate :wink: )

stevecoh1, 2024-04-25:
1. Do all Pi-5s have a power button?  I don't think mine does.  It's a Canakit Pi-5 with the aluminum case. Am I missing something?
2. What is the "shutdown state" and why does it occur?  Besides pressing the power button (if there is one) is there another way to get it out of that state? In my case I left it on, idle for a couple of hours, and found it in this state.

lurch, 2024-04-26:
@stevecoh1 Questions like that should be asked on the [forums](https://forums.raspberrypi.com/), not on Github.
But yes, every Pi 5 has a power-button (see https://www.raspberrypi.com/products/raspberry-pi-5/ ), but perhaps not all Pi 5 cases make it externally-accessible? :shrug: