raspberrypi/documentation #3541: Document power draw of the official case fan and the active cooling fan

C0rn3j, 2024-02-17:
As per https://github.com/raspberrypi/rpi-eeprom/issues/545#issuecomment-1949426226, power draw is shared between USB and the fan header, making knowing the power draw of the fans important.

The documentation is missing [here for the fan case](https://www.raspberrypi.com/documentation/computers/raspberry-pi-5.html#fan-case) and [here for the active cooler](https://www.raspberrypi.com/documentation/computers/raspberry-pi-5.html#active-cooler)

C0rn3j, 2024-04-21:
@aallan @LizUpton @nathan-contino 

Would be nice if the stale bot marked thing stale instead of stale+spam+autoclose.

stevedavies-rp, 2026-10-01:
Thank you for reporting this. This has now been documented and will be published very soon.

closed 2026-10-01 by stevedavies-rp