raspberrypi/documentation #3542: Datasheet for ML2020 RTC battery for Pi5 is missing

abaumgaertner, 2024-02-20:
Please provide the datasheet for the 
ML2020 - https://www.raspberrypi.com/products/rtc-battery/ 
here: https://datasheets.raspberrypi.com

voltage, capacity, trickle charge specs, temp. operating range and temp limits

closed 2024-02-20 by aallan

JamesH65, 2024-02-20:
Note, do not trickle charge - see the datasheet.

abaumgaertner, 2024-02-20:
> Note, do not trickle charge - see the datasheet.

my information about the trickle charge method is based on this article:
https://www.raspberrypi.com/news/how-to-use-the-real-time-clock-functionality-in-raspberry-pi-5-magpimonday/
and was also mentioned in earlier versions of the documentation
https://github.com/raspberrypi/documentation/commit/5c7c0aabb23f2b22cd9c2a1e53392436ffd104d9

there is also forum thread about the issue regarding ML2020 and trickle charge
https://forums.raspberrypi.com/viewtopic.php?t=363662

@JamesH65 charging method is clarified now - thank you
https://www.raspberrypi.com/documentation/computers/raspberry-pi-5.html#enabling-battery-charging
The RTC is equipped with a constant-current (3mA) constant-voltage charger.

Please add the datasheet to https://datasheets.raspberrypi.com)https://datasheets.raspberrypi.com/

JamesH65, 2024-02-20:
Yes, that article needs to be updated, following a change to the RTC documentation.

I am not sure we can or want to add third-party datasheets to our datasheets folder, there may be copyright issues, and there is the constant problem of keeping things updated.

aallan, 2024-02-20:
No. We can't (shouldn't be) host(ing) third party datasheets. Setting aside copyright and other issuesWe wouldn't want others hosting potentially outdated copies of our datasheets. We shouldn't be doing that to others!