raspberrypi/documentation #3556: Make the `$ ` at the beginning of interactive code/console snippets uncopyable.

nathan-contino, 2024-03-11:
As per the suggestion here: https://github.com/raspberrypi/documentation/pull/3549#discussion_r1517429422

There are advantages to prefixing interactive CLI snippets with the `$` symbol. However, it can decrease copy/paste friendliness and present an obstacle to new users. We should:

* omit the `$ ` prefix when you click the the "copy to clipboard" button on code snippets
* [stretch goal] make the `$ ` prefix unselectable in CSS styling such that users physically cannot copy and paste it when they grab the snippet manually (without clicking the 'copy to clipboard' button)

Right now we manually add the `$ ` prefix at the beginning of these snippets. We should consider *automatically* adding the prefix to the beginning of `console` type snippets.

nathan-contino, 2024-03-11:
For example, see the code snippets in this section: https://www.raspberrypi.com/documentation/computers/configuration.html#change-a-users-password

Note that interactive code snippets are sometimes multi-line, but those lines should almost always get copy/pasted one at a time. For multiline snippets, we want a `$ ` prefix on every line, and either _no_ "copy to clipboard" button (because you would never want to copy all the lines at once) or a "copy to clipboard" button for each line individually (which sounds slightly awkward). You could possible break up multiline snippets into multiple one-line snippets either automatically or manually (and then throw a warning if someone creates a multiline `console` snippet in the future).

aallan, 2024-03-11:
> omit the $  prefix when you click the the "copy to clipboard" button on code snippets

This is something I want.

> We should consider automatically adding the prefix to the beginning of console type snippets.

This sounds awfully fragile, and means combing the documentation for every code block and deciding whether it is or isn't a command line example.

nathan-contino, 2024-05-12:
Too fast!