raspberrypi/documentation #3589: Failure to build: 403 (Forbidden) from sass-embedded gemfile using bundle

MicrosoftWorks, 2024-04-12:
Error when using bundle, RPiOS, RPi 5:

`Bundler::HTTPError:Could not download gem from https://rubygems.org/ due to underlying error <bad
response Forbidden 403 ([https://rubygems.org/gems/sass-embedded ... -linux.gem](https://rubygems.org/gems/sass-embedded-1.72.0-aarch64-linux.gem))>`

Quoted from a user with the same issue on the forums, [https://forums.raspberrypi.com/viewtopic.php?t=368347](url)

> which looks like an error on rubygems.org.
>However, I've successfully installed sass-embedded from as described here https://rubygems.org/gems/sass-embedded.
>Has anyone else experienced this? Suggestions?

The issue seems to be related to bundler not adding the proper suffix to the link, [https://rubygems.org/gems/sass-embedded-1.72.0-aarch64-linux.gem](url) wherein rubygems.org  an additional descriptor needs to be added after, `...aarch64-linux [here].gem`i.e. [https://rubygems.org/gems/sass-embedded-1.69.5-aarch64-linux-gnu.gem](url) which actually provides a gem.

I'll be happy to provide any additional info.

aggo15, 2024-04-16:
Can confirm. I'm having the same issue as described by OP. I'm building in a docker container from image `debian:latest` and hit by the same error message.

closed 2024-06-30 by github-actions[bot]

nathan-contino, 2024-07-01:
Is this still an issue? We can reopen and investigate, but unless I hear that this is still creating problems, I'll side with stalebot and allow this one to close.