raspberrypi/documentation #3591: DebugProbe version needs updating

lurch, 2024-04-16:
https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#updating-the-firmware-on-the-debug-probe says "The current version of the Debug Probe firmware is version 1.0.3.", but [debug probe v2.0.1](https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.0.1) was recently released.
So either the version needs updating, or maybe that NOTE should be tweaked to not actually mention any specific version number? :thinking:

nathan-contino, 2024-04-16:
Already fixed! https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/microcontrollers/debug-probe/updating-firmware.adoc
... I just need to roll out the changes.

lurch, 2024-04-16:
Hah, whoops!

closed 2024-04-16 by lurch

lurch, 2024-04-17:
Website-docs have now been updated :+1: