raspberrypi/documentation #3593: Audio configuration docs are incorrect/incomplete

dangarbri, 2024-04-17:
I've been struggling to make the Codec Zero work in a headless environment. I don't know enough about alsa to make this work right...

I followed the instructions, and the codec zero hat works in the desktop environment when I right click the sound settings and select the codec zero, but I can't get it to work in the lite/headless version of the OS.

Specifically, this section in the docs doesn't work on the raspberry pi 4 model B.

https://github.com/raspberrypi/documentation/blob/4ef432ca21ffe4c0aa3da3276ad39665f7e6a95b/documentation/asciidoc/accessories/audio/configuration.adoc?plain=1#L126-L139

It will cause a ton of headaches:
```
root@raspberrypi:~# speaker-test

speaker-test 1.2.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Channels count (1) not available for playbacks: Invalid argument
Setting of hwparams failed: Invalid argument
```

Also, trying to restore the alsa state doesn't seem to work either

https://github.com/raspberrypi/documentation/blob/4ef432ca21ffe4c0aa3da3276ad39665f7e6a95b/documentation/asciidoc/accessories/audio/configuration.adoc?plain=1#L87-L91

```
root@raspberrypi:~# sudo alsactl restore -f /home/pi/Pi-Codec/Codec_Zero_OnboardMIC_record_and_SPK_playback.state 
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
root@raspberrypi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Zero [RPi Codec Zero], device 0: Raspberry Pi Codec Zero HiFi da7213-hifi-0 [Raspberry Pi Codec Zero HiFi da7213-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@raspberrypi:~# 
```

lurch, 2024-04-17:
@nathan-contino I'll take this one, because I was last looking at all this about a month ago.

dangarbri, 2024-04-17:
I appreciate any help 🙂

I'm available to test anything. I think it's odd that selecting the audio device in the desktop version works. I don't know what that's doing differently.

dangarbri, 2024-04-18:
After tinkering some more, I think (maybe) the only issue is this.

 ```
 pcm.!default { 
         type hw 
         card Zero 
 } 
 ```

The `alsactl` command prints errors, but it still does what it's supposed to do. The codec zero won't work without running that.

So it's just setting the default device on the command line that has issues. On the desktop OS where I have everything working, if I run `speaker-test` without any arguments, it plays the pink noise as expected. But if I try to run speaker-test by manually specifying the device, it doesn't work:

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 3: Zero [RPi Codec Zero], device 0: Raspberry Pi Codec Zero HiFi da7213-hifi-0 [Raspberry Pi Codec Zero HiFi da7213-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ speaker-test -D hw:3,0

speaker-test 1.2.8

Playback device is hw:3,0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Channels count (1) not available for playbacks: Invalid argument
Setting of hwparams failed: Invalid argument
pi@raspberrypi:~ $ 
```

nathan-contino, 2024-06-24:
Still on the radar, just haven't had the time to sit down and figure this out recently.

stevedavies-rp, 2026-09-16:
Thank you for raising this issue. The documentation is now being updated and will be live very soon.

closed 2026-09-16 by stevedavies-rp