raspberrypi/documentation #3635: Can't flash eMMC of CM4 using another Pi 4

ll-O-ll, 2024-05-14:
So I've been trying to flash the eMMC of my CM 4 using another Raspberry Pi 4. I've used the steps outlined at https://www.raspberrypi.com/documentation/computers/compute-module.html#steps-to-flash-the-emmc 

Here's what I did:

Connect jumper cable on J2 to disable eMMC boot on the CM 4. 
Connect J11 USB slave to Pi 4
Power Pi 4
Power CM 4

Here's a picture of my setup: 

[image]

I expect the CM 4 to appear as a device on the Pi 4 but I see nothing :(

Any steps I'm missing here??

lurch, 2024-05-14:
Have you built and run `rpiboot` ? https://www.raspberrypi.com/documentation/computers/compute-module.html#building-rpiboot-on-your-host-system

ll-O-ll, 2024-05-16:
Yes I did this... Rpiboot stuck on Waiting for BCM2835/6/7/2711

dp111, 2024-05-16:
Have you tried a different USB cable?

ll-O-ll, 2024-05-28:
Tried a different cable it works now. New problem though. It freezes my Pi 4 machine and all peripherals don't work (mouse keyboard). I rebooted both pis and the same thing happened. Here's a snapshot of where i'm at:
[image]

The CM4's green light turns on as well... seems to me it exits boot mode

ll-O-ll, 2024-05-28:
So I tried the secure boot method... it got me further but I still can't see the disk mount when I run `sudo lsblk`

closed 2025-06-11 by lurch