raspberrypi/documentation #3671: Adjust the readability

qtop1, 2024-05-26:
Hi,

Checking the documentation and would like to alter the font  and sizing of documentation to make more accessible to users  with vision impairments and dyslexic users. As the font is quite dense.

Please let me know if this is something I can adjust/ contribute to this project?

nelliemckesson, 2024-05-26:
@qtop1 this is definitely something that's important to us. I think a lot of browsers come with some built-in tools for this sort of thing, but I'll raise the issue with the web and design teams and see what everyone says. Any additional detail you can give about the specific issues you've identified would be greatly appreciated!

qtop1, 2024-06-01:
Hi @nelliemckesson , 

In terms of more details :

The main issue is the color contrast between the text and background. The contrast is quite light, making it difficult for users with vision impairments. Consider changing to a darker color.

As for the font, the text is very close together and thin, which is not dyslexia-friendly. As someone who is dyslexic, I find it makes reading more difficult and time-consuming, especially when comprehension is affected. The font doesn't need to be bold, as that can also be hard to read, but it should have more spacing between the letters.

nathan-contino, 2024-06-03:
It would be helpful to know which text you're referring to. I just checked, and our body, header, ToC, and tab text all meet WCAG standards for contrast.

As far as the font is concerned, our body text uses Roboto. I'm not sure I'd call it particularly "close together and thin" compared to a lot of fonts on the Internet, but I can see how it's less readable than a font specifically designed for dyslexic readers. Unfortunately I don't think using a (specifically designed for) dyslexia-friendly font is a typical best practice, likely because those fonts usually don't look very appealing aesthetically.

Let me know if you have a specific solution in mind for the font, but I just checked the accessibility report on our site and it seems like we're addressing all of the basics.

nelliemckesson, 2024-06-03:
@qtop1 I chatted with our web team about this and we're looking at swapping the font out for a slightly heavier weight (not bold, but it's currently using one of the thinner weights), as well as bumping up the size a little bit. In terms of the font itself, most browsers allow you to change your font settings -- it's a bit of a balancing act on our end to find settings that work for everyone, so I encourage you to look into what you can do through your browser in that regard.

closed 2024-08-11 by github-actions[bot]