raspberrypi/documentation #3692: Linked documentation for python's pip install  externally-managed-environment has missing Anchor tag

DeeJay, 2024-06-05:
Linked documentation for python's pip install  externally-managed-environment has missing Anchor tag

Text emitted by `pip install fred` includes 

> For more information visit http://rptl.io/venv

That short URL expands to 

[https://www.raspberrypi.com/documentation/computers/os.html#python-on-raspberry-pi](https://www.raspberrypi.com/documentation/computers/os.html#python-on-raspberry-pi)

But the **#python-on-raspberry-pi** anchor tag has been removed from the documentation.

So the link unhelpfully rests at [https://www.raspberrypi.com/documentation/computers/os.html#introduction](https://www.raspberrypi.com/documentation/computers/os.html#introduction)

The correct target URL should be [https://www.raspberrypi.com/documentation/computers/os.html#install-python-libraries-using-pip](https://www.raspberrypi.com/documentation/computers/os.html#install-python-libraries-using-pip)

lurch, 2024-06-05:
Looks like this was inadvertently broken by a header-rename in #3665

nathan-contino, 2024-06-05:
Thanks for bringing this up! It's tough to keep implicit anchor points consistent when I'm cleaning up sections of the docs. Fixing this shortly.

closed 2024-06-05 by nathan-contino