raspberrypi/documentation #3711: Flashing the Compute Module eMMC

mrawson-mestek, 2024-06-07:
There is a minor documentation issue in the section on "Flashing the Compute Module eMMC" at:

https://www.raspberrypi.com/documentation/computers/compute-module.html

If you are following the steps for the CM4 on Linux, the first step says "Do not power up yet.", but there is nothing that tells you to apply power before (or after) running RPIBoot.

IMO, the section should be changed to something like...

Writing to the eMMC (Linux)

* Apply 12VDC power to the IO Board.
* Run rpiboot
* Plug the micro usb cable into the host machine.

After rpiboot completes, you will see a new device appear; this is commonly /dev/sda on a Raspberry Pi but it could be in another location such as /dev/sdb, so check in /dev/ or run lsblk before running rpiboot so you can see what changes.

You now need to write a raw OS image (such as Raspberry Pi OS) to the device. Note the following command may take some time to complete, depending on the size of the image (change /dev/sdX to the appropriate device):

nathan-contino, 2024-06-07:
Already completely reworked this in https://github.com/raspberrypi/documentation/pull/3690. That section doesn't even contain the text "Do not power up yet" any more because I felt the same as you. I also added explicit instructions on each platform to power up before running `rpiboot`.

If you're still seeing the old version of the page, try opening it in a private window or flushing your browser cache. Thanks for the feedback, it's nice to know I beat someone to the punch for once 😄

closed 2024-06-07 by nathan-contino

mrawson-mestek, 2024-06-07:
😄I actually sent this in via a contact form or email (I forget) a week or two ago.  Today I got an email from Nicola Early requesting that I raise it as an issue and I didn't even look to see if it still existed... oops.

Thanks for fixing it, hopefully it will save someone else a little confusion.