raspberrypi/documentation #3755: Codec Zero HAT cannot load alsamixer state & also gives read error

cubedmeatgoeshere, 2024-07-01:
I have a Codec Zero on a Pi Zero 2 W running the latest PiOS Lite 64bit Bookworm in headless configuration

`/proc/device-tree/hat/name:hat
/proc/device-tree/hat/product:Pi-CodecZero
/proc/device-tree/hat/product_id:0x0013
/proc/device-tree/hat/product_ver:0x0001
/proc/device-tree/hat/uuid:e15c739c-877d-4e29-ab36-4dc73c21127c
/proc/device-tree/hat/vendor:IQaudIO Limited www.iqaudio.com
`
I've been following the standard documentation on how to configure this HAT via https://www.raspberrypi.com/documentation/accessories/audio.html

Steps taken:
edited /boot/firmware/config.txt:
`
#dtparam=audio=on

dtoverlay=
dtoverlay=rpi-codeczero
`

Added 
`pcm.!default {
        type hw
        card Zero
}`
to .asoundrc

However these alsactl restore commands always fail:
`alsactl restore -f ~/Pi-Codec/Codec_Zero_StereoMIC_record_and_HP_playback.state 
alsactl restore -D Zero -f ~/Pi-Codec/Codec_Zero_StereoMIC_record_and_HP_playback.state 
`
`alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
`

The device seems like it's loaded:
` arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Zero [RPi Codec Zero], device 0: Raspberry Pi Codec Zero HiFi da7213-hifi-0 [Raspberry Pi Codec Zero HiFi da7213-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`
and alsamixer shows the device and I can play around with input levels but I don't see how I can turn on/off inputs/outputs with spacebar or any other key

And when I try using the device with `arecord -D hw:0,0 -f cd test.wav`, I get this error
`Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: pcm_read:2221: read error: Input/output error
`

I've tried many permutations of configurations including disabling HDMI via `#dtoverlay=vc4-kms-v3d`

This may be related to the following other issues:
https://github.com/raspberrypi/documentation/issues/3593
https://github.com/raspberrypi/documentation/issues/3292#issuecomment-1911443719

Also this may be specific to a recent update and/or Bookworm because I was using this HAT w/ this very Pi on 64 bit Lite Bullseye perfectly fine a few months ago

nathan-contino, 2024-07-01:
Thanks for reporting this. I don't know how to fix this personally, but I'll try raising it internally and report back if we get anywhere. I strongly suspect that Bookworm broke something here, but I'm not sure what.

learmj, 2024-09-30:
Hi

> I have a Codec Zero on a Pi Zero 2 W running the latest PiOS Lite 64bit Bookworm in headless configuration

> However these alsactl restore commands always fail: `alsactl restore -f ~/Pi-Codec/Codec_Zero_StereoMIC_record_and_HP_playback.state ` `alsactl restore -D Zero -f [...]

This is probably a harmless error. See https://github.com/raspberrypi/documentation/pull/3855
 
> And when I try using the device with `arecord -D hw:0,0 -f cd test.wav`, I get this error `Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo arecord: pcm_read:2221: read error: Input/output error`

This likely means there is a low level hardware issue. Examining dmesg may show other kernel errors. I would examine the output of ```aplay -l``` after this error to list the supported devices in the system. ```hw:X,Y``` means ```hardware:<card number>,<device number>``` and it may change between boots depending on devices pluged/unplugged. Also note that ```hw:0,0```  means the default device on the default sound card [1].

[1] https://www.alsa-project.org/main/index.php/Asoundrc

closed 2024-12-15 by github-actions[bot]

cubedmeatgoeshere, 2025-02-06:
> This likely means there is a low level hardware issue. Examining dmesg may show other kernel errors. I would examine the output of `aplay -l` after this error to list the supported devices in the [...]

arecord and aplay show card and device
` arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: Zero [RPi Codec Zero], device 0: Raspberry Pi Codec Zero HiFi da7213-hifi-0 [Raspberry Pi Codec Zero HiFi da7213-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

dmesg gives this when attempting an arecord:
`[  110.126624] bcm2835-i2s 3f203000.i2s: I2S SYNC error!`
Which seems unhelpful.

I ultimately took this advice of these two users:
https://forums.raspberrypi.com/viewtopic.php?t=306800
https://forums.raspberrypi.com/viewtopic.php?t=375686
And did the following:
1) installed Bullseye
2) did NOT add dtoverlay=rpi-codeczero to config.txt and only commented out #dtparam=audio=on

Interestingly the card/device now shows up as 
`IQaudIOCODEC [IQaudIOCODEC], device 0: IQaudIO CODEC HiFi v1.2 da7213-hifi-0 [IQaudIO CODEC HiFi v1.2 da7213-hifi-0]`
instead of what I'd previously reported above (`Zero [RPi Codec Zero], device 0: Raspberry Pi Codec Zero HiFi da7213-hifi-0 [Raspberry Pi Codec Zero HiFi da7213-hifi-0]`). 

Success:
`sudo alsactl restore -d IQaudIOCODEC -f ./Pi-Codec/Codec_Zero_StereoMIC_record_and_
HP_playback.state
alsactl: set_controls:1479: device='hw:2', doit=0
alsactl: set_controls:1491: card-info-id: 'IQaudIOCODEC'
alsactl: set_controls:1517: maxnumid=97
alsactl: set_controls:1535: result code: 0
alsactl: set_controls:1479: device='hw:2', doit=1
alsactl: set_controls:1491: card-info-id: 'IQaudIOCODEC'
alsactl: set_controls:1517: maxnumid=97
alsactl: set_controls:1535: result code: 0`

and `arecord` successfully gave me the microphone recording that I'd been wanting for a year...

However back on Bookworm with the SAME config changes:
`$ sudo alsactl restore -D IQaudIOCODEC -f Codec_Zero_StereoMIC_record_and_HP_playback.state
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2`
Womp womp...

This makes me believe that dtoverlay=rpi-codeczero was misconfiguring my device, and also that Bookworm has lost the original configuration that enables my legacy IQaudio device. At the very least I recommend the docs be updated to reflect this.