raspberrypi/documentation #3764: Raspberry Pi Connect | WINDOWS

Bra1nsen, 2024-07-04:
Dear Developer Team,

I have created a Raspberry Pi account and am logging in via the website from a Windows PC. How can I add a Raspberry Pi device?

Is it currently only possible to access a Pi from another Pi?

[image]

Best regards,
Paul

lurch, 2024-07-04:
1. Please see https://www.raspberrypi.com/documentation/services/connect.html#link-a-raspberry-pi-with-a-raspberry-pi-id
2. Please ask questions like this at https://forums.raspberrypi.com/viewforum.php?f=167 rather than on GitHub.

closed 2024-07-04 by lurch

Bra1nsen, 2024-07-04:
So could you just confirm thats not yet working for Windows? Only Pi-to-Pi?

lurch, 2024-07-04:
The Connect website works fine on Windows, but you need to use the Pi itself to register the Pi with your Connect account. But once you've done that you can access the Pi from Windows.
See https://www.raspberrypi.com/news/raspberry-pi-connect/

Bra1nsen, 2024-07-04:
Thanks alot Lurch. Last question.🙏 Is it allowed for companies to use Raspberry Connect or does it require a license?

Best, 
Paul