raspberrypi/documentation #3809: Design files for the Pico 2 appear to be an RP2350 Minimal Design

sabogalc, 2024-08-11:
It appears that the design files provided at the below link are for a RP2350 minimal design rather than for a Pico 2
https://datasheets.raspberrypi.com/pico/RPi-Pico-2-PUBLIC-20240708.zip

See also #3807

JamesH65, 2024-08-11:
Which part(s) of the docs links to that file?

sabogalc, 2024-08-11:
The below two links have this link listed as Cadence Allegro files for the Pico 2

https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#pinout-and-design-files
https://www.raspberrypi.com/documentation/microcontrollers/silicon.html#rp2350-based-boards

nathan-contino, 2024-08-20:
Removed the references for now; working on getting the proper files soon.

closed 2024-08-20 by nathan-contino

gemarcano, 2025-03-16:
This is also the case for the Pico 2 and Pico 2 datasheets:
https://datasheets.raspberrypi.com/picow/pico-2-w-datasheet.pdf
https://datasheets.raspberrypi.com/pico/pico-2-datasheet.pdf

In section 1.1 they both point to:
https://datasheets.raspberrypi.com/pico/RPi-Pico-2-PUBLIC-20240708.zip

Which just contains the RP2350 minimal design instead of the layout for the Pico 2 or Pico 2 W.

--

Related (should I open a new documentation issue?), the STEP file in section 1.1 for the Pico 2 W is identical to the Pico 2 (which is correct as far as I can tell). For basic prototyping I think this is OK, but it'd be nice if it looked a little closer to the Pico 2 W.