raspberrypi/documentation #3814: missing tensorflow-lite package-config file after installation

LucasLvbj, 2024-08-13:
where: doc of "Post-Processing with TensorFlow Lite" -> "Prerequisites"
Issue description:

- after download and install a version that exports the API from [lindevs.com], and try to recompile rpicam-apps with TensorFlow Lite support]. 
- at step: "meson setup build -Denable_libav=enabled -Denable_drm=enabled -Denable_egl=enabled -Denable_qt=enabled -Denable_opencv=enabled -Denable_tflite=enabled", got error [ERROR: dependency "tensorflow-lite" not found]
- after google, this seems solved in https://github.com/raspberrypi/rpicam-apps/issues/622, but not reflected in doc. Suggest to update doc.

lurch, 2024-08-13:
ping @davidplowman and @naushir

closed 2024-10-20 by github-actions[bot]

lurch, 2024-10-22:
@LucasLvbj Is this still an issue, or did it get resolved as part of the recent AI Kit / AI Camera documentation updates?