raspberrypi/documentation #3821: Add maximum bare-board power consumption to table

clark800, 2024-08-17:
The power consumption table [here](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/power-supplies.adoc#typical-power-requirements) shows "Typical bare-board active current consumption", but it would also be helpful to know the "maximum bare-board current consumption".

The recommended power supply numbers in the table seem to be padded to accommodate peripherals, which means there is currently is no reliable way to determine if a smaller power supply would be sufficient for a particular use case, such as a headless server with no peripherals. Knowing the "maximum bare-board current consumption" would allow users to calculate an optimal power supply by adding this number to the current consumption required for their particular peripheral setup.

The "stress" numbers in the following table give some indication, but they don't currently cover all models, they aren't bare-board numbers, and it's not clear if they are maximal in the sense of full utilization of all cores plus WiFi plus bluetooth etc simultaneously.

Bossdell113, 2024-10-04:
Hello.  I am taking a software engineering class and I was wondering if I could claim this issue for a project I am working on.  If I am understanding this correctly the table only shows the "typical" bare board consumption when it should also show the maximum bare board consumption.  If I am correct I have two questions.  1. How can I figure out what the maximum consumption for these models are?  2. When would you want this done?  I'm sure I can get this done but I'm probably going to need about 3, 4, maybe even 5 weeks for this.  I have never used a raspberry PI so this is all completely new to me, but since this is just documentation related I feel like this is more in my comfort zone.

JamesH65, 2024-10-04:
It is entirely dependent on the use case. For example, normal desktop use will be approximately one figure, but if you then do a video decode it will be more (and what format is the video in, that also makes a difference), but if you then add some pathological processing case it will be even more. Add a camera? Even more. Now doing a big data transfer over ethernet? More again.

The only reliable way of determining how much power the device will use in your own particular use case is to test it.

Hence we use typical values.

But it would be interesting to see results for all of these cases but the table would be complex!

clark800, 2024-10-04:
Thanks for the responses here.

@JamesH65 The idea behind listing the *maximum* consumption is that this would not be dependent on the use case since it is the maximum over all possible use cases. And "bare-board" means this is well-defined and not dependent on peripherals like a camera - the only physical connection to the board is the power supply. We can also specify that this is assuming no overclocking or other custom tweaks.

Ideally this could be calculated by adding the maximum consumption of each component on the board and perhaps adjusting upwards for losses or downwards for any components that can't operate simultaneously.

@Bossdell113 I think probably only Raspberry Pi engineers could really address this authoritatively.

lurch, 2024-12-09:
I'm going to go ahead and just close this. We don't encourage users to "calculate an optimal power supply", we just advise them to buy one of our [official power supplies](https://www.raspberrypi.com/products/power-supply/).
If your needs are more exotic than this, then you'll need to do testing/measuring of your own specific requirements, as James has already said.

closed 2024-12-09 by lurch