raspberrypi/documentation #3822: Invalid link to firmware/overlays README file.

scotty3785, 2024-08-19:
[****](https://github.com/raspberrypi/documentation/blame/d96c5376b304b89ef5620e42969a61c727c3c871/documentation/asciidoc/computers/configuration/device-tree.adoc#L827)

This documentation currently says "Please refer to the [README](https://github.com/raspberrypi/firmware/blob/master/boot/firmware/overlays/README) file found alongside the overlay .dtbo files in /boot/firmware/overlays. It is kept up-to-date with additions and changes." 

The link to the overlays README returns a 404 error.

lurch, 2024-08-19:
Ahhh, looks like the link ought to be https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README - perhaps things got rearranged? ping @popcornmix

nathan-contino, 2024-08-19:
Thanks! Fixing this now...

nathan-contino, 2024-08-19:
https://github.com/raspberrypi/documentation/pull/3823

closed 2024-08-19 by nathan-contino