raspberrypi/documentation #3831: Photo for pico without the JST-SH 1mm debug socket.

jmnc2, 2024-08-31:
This is about the pico and debug-probe/getting-started document, see link below.

Be nice if at least 1 of the photos, diagrams or video showed the wiring with the wire colours for the boards without the JST 1mm socket.

I mean, do the people with the H boards even really need the photo to show the debug socket, as I think it can only go one place anyway?

[debug-probe.html#getting-started](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#getting-started)

P.S. This added to the no marking on the top of the board, I find very vexing. ;(

JamesH65, 2024-08-31:
That is a very good point, I had no idea that wasn't in the docs. @nathan-contino  This really ought to show the wiring for both types of Pico (with/without JST). The datasheet is the same and would also need updating.

nathan-contino, 2024-08-31:
On it, thanks!

lurch, 2026-01-18:
For people looking for this info in the meantime, note that the Getting Started PDF (which is unfortunately also missing a non-PicoH diagram) available at https://pip.raspberrypi.com/categories/610-raspberry-pi-pico says: 

> NOTE: If you have a non-H Pico, Pico 2 or Pico W (without a JST-SH connector) you can still connect it to a Debug Probe.
Solder a male connector to the SWCLK, GND, and SWDIO header pins on the board. Using the alternate 3-pin JST-SH
connector to 0.1-inch header (female) cable included with the Debug Probe, connect to the Debug Probe "D" port.
Connect SWCLK, GND, and SWDIO on the Pico or Pico W to the SC, GND, and SD pins on the Debug Probe, respectively.