raspberrypi/documentation #3836: Global shutter multiple camera synchronization documentation

justinblaber, 2024-09-04:
Was taking a look at: [https://www.raspberrypi.com/documentation/accessories/camera.html](https://www.raspberrypi.com/documentation/accessories/camera.html)

Was wondering if there is more information on the roles of `XHS`, `XVS`, and `XTRIG` in synchronized capture? Coming from USB machine vision cameras usually a single GPIO pin is used to start capture at the same time for multiple cameras, so was trying to find more info on the specific roles of `XHS`, `XVS`, and `XTRIG`. `XTRIG` and `XVS` in particular seem very similar in description (synchronizes start of exposure?) and `XHS` seems like it should be specifically for rolling shutter (synchronizes start of each line?) although the docs mention it's also used for GS cameras which seems contradictory. I've googled around but it's difficult to find info on this topic. If anyone has any info or docs they could link that would be great.

lurch, 2024-09-05:
ping @davidplowman and @naushir

naushir, 2024-09-05:
``XTRIG`` is a trigger used for external flash strobing whereas ``XVS``/``XHS`` are used as sync pulses.  Sadly we cannot share datasheets as they are under an NDA with Sony.

closed 2024-11-17 by github-actions[bot]