raspberrypi/documentation #3844: Instructions for compiling device trees no longer work

ali1234, 2024-09-10:
https://www.raspberrypi.com/documentation/computers/compute-module.html#device-tree-source-and-compilation

To reproduce:

1. Make a fresh install of Raspberry Pi OS Lite 64 bit
2. `sudo apt install git bc bison flex libssl-dev make device-tree-compiler
3. `git clone --depth 1 https://github.com/raspberrypi/linux`
4. `cd linux/arch/arm/boot/dts/overlays`
5. `dtc -@ -Hepapr -I dts -O dtb -o vc4-kms-v3d-pi5.dtbo vc4-kms-v3d-pi5-overlay.dts`

Result:

```
Error: vc4-kms-v3d-pi5-overlay.dts:3.1-2 syntax error
FATAL ERROR: Unable to parse input tree
```

Now try the version of dtc built by the kernel itself:

1. `cd ~/linux`
2. 'KERNEL=kernel_2712`
3. `make bcm2712_defconfig`
4. `make dtbs`
5. `scripts/dtc/dtc -@ -Hepapr -I dts -O dtbo -o vc4-kms-v3d-pi5.dtbo arch/arm/boot/dts/overlays/vc4-kms-v3d-pi5-overlay.dts`

Result:

```
Error: arch/arm/boot/dts/overlays/vc4-kms-v3d-pi5-overlay.dts:3.1-2 syntax error
FATAL ERROR: Unable to parse input tree
```

lurch, 2024-09-10:
ping @pelwell

pelwell, 2024-09-11:
The OP has missed the point - the comments are about the simple example presented in the docs (https://www.raspberrypi.com/documentation/computers/configuration.html#part2.1 - it's referred to as `1st-overlay.dts`, and the third line is `/plugin/;`.

The problem with trying to build `vc4-kms-v3d-pi5.dtbo` by hand is that the overlays in the kernel tree tend to assume that the C preprocessor (`cpp`) has been run on the source file to expand `#include`s and perform macro substitution. This isn't a problem because the kernel's Makefiles take care of that, so `make ... dbs` is the preferred way to build them.

For an example of the steps needed, take a look at `overlaycheck` in our `utils` repo. This (https://github.com/raspberrypi/utils/blob/master/overlaycheck/overlaycheck#L804) is a function that runs `cpp` before `dtc`, using the definition of `@cpp_cmd` created here (https://github.com/raspberrypi/utils/blob/master/overlaycheck/overlaycheck#L148).

ali1234, 2024-09-11:
If I've missed the point, it's because the point is not covered in the documentation.

pelwell, 2024-09-11:
Perhaps a note such as:
> Overlays and `.dts` files found in the kernel source tree should be compiled using the kernel's makefiles, i.e. `make ARCH=arm ... dtbs` or `make ARCH=arm64 ... dtbs`. Running `dtc` manually is likely to lead to errors because of the use of `cpp` directives in the source.

lurch, 2024-09-11:
:point_up: @nathan-contino

pelwell, 2024-09-20:
@ali1234 The [utils](https://github.com/raspberrypi/utils) repo has just gained another command - [kdtc](https://github.com/raspberrypi/utils). It's a wrapper around `dtc` that runs the inputs through `cpp`. It also has a _simple_ mode that allows you to write `kdtc <input> <output>` and have it do the right thing.

nathan-contino, 2024-11-24:
we'll get there soon

stevedavies-rp, 2026-09-24:
Thank you for this feedback and for your patience. Appropriate documentation updates are now underway.

closed 2026-09-24 by stevedavies-rp